LibrePCB/LibrePCB #739: Schematic Editor : raise a warning if one net is connected to only one pin

doom-fr, 2020-07-31:
##### SUMMARY
I usually use nets to connect pins in schematic to make it more clear.
Unfortunately, there's a risk to miss an unconnected net.
(unfortunately, I experienced that !)

##### SOLUTION
Raise a warning if one net is connected to only one pin.

ubruhin, 2020-08-01:
Hmm I don't understand, actually the ERC already raises a warning in this case:

[image]

Or what do you mean exactly?

doom-fr, 2020-08-01:
Unfortunately, I use a 'schematic only' component to set the net : 

[image]

And like that, it does not raise eror or warning.
So I think that 'schematic only' component should not be counted as real connection !

PS : maybe you can also add connection port(s) as native in LibrePCB  [image]

(I know, this should be in another issue !)

dbrgn, 2020-08-01:
Hm, why don't you just set the net using a net label?

doom-fr, 2020-08-01:
It's more clear if it was input, output or input/output (I have the 3 variants), and I found it more visible.
And I used it before in Protel !

5n8ke, 2020-08-01:
Ignoring schematic-only devices for determining, whether a signal is connected to more than one pin, would also show a warning when only one pin is connected to a VCC or GND device. This is especially useful when having multiple different supply voltages.

doom-fr, 2020-08-01:
??
Sorry but do not understand your argument.
I even found that this reinforced my position for this issue !

As this kind of schematics makes no sense but do not raise error/warning : 
(here R1 is in fact connected to anything ... (place the device on the board and you will see))

[image]

doom-fr, 2020-08-01:
> Ignoring schematic-only devices for determining, whether a signal is connected to more than one pin, would also show a warning when only one pin is connected to a VCC or GND device. This is especially useful when having multiple different supply voltages.

??
Sorry but do not understand your argument.
I even found that this reinforced my position for this issue !

As this kind of schematics makes no sense but do not raise error/warning : 
(here R1 is in fact connected to anything ... (place the device on the board and you will see))

[image]

5n8ke, 2020-08-02:
> ??
> Sorry but do not understand your argument.
> I even found that this reinforced my position for this issue !
> 
> As this kind of schematics makes no sense but do not raise error/warning : 
> (here R1 is in fact connected to anything ... (place the device on the board and you will see))

Exactly my point! I was making an argument for changing the current behavior.

Currently schematic-only devices are not ignored, when determining the count of connected pins to a NetSignal

doom-fr, 2020-08-02:
> Exactly my point! I was making an argument for changing the current behavior.
> Currently schematic-only devices are not ignored, when determining the count of connected pins to a NetSignal
Ok. Sorry. So I just misunderstood what you said.
And so we both agree !

PR #745 (2020-08-07): ERC: Disregard schematic-only components in net pin count
Only components that are not schematic only are counted when determining if a NetSignal is connected to less than two pins.

- Fixes #739
closed 2020-08-07 by ubruhin