LibrePCB/LibrePCB #741: Add a way to duplicate project from the software

miniwark, 2020-08-02:
##### SUMMARY

The software do not allow to duplicate projects, either by a "Duplicate" or "Save As..." features. 

- This could be useful in case of "projects as a template": for example for boards with the same form factor (like Arduino shields)
- "Manual forking" to keep v1/v2/v3 ect. version of a board in there own project

##### SOLUTION

It's already possible to work around this outside the software by: 
- copy/paste a project folder
- rename the folder
- rename the main `.lpp` file
- edit the main title of the Readme file
- edit the name in the metadata file (preferably from menu: Project > Promerties > Name)

A "save As..." or "Duplicate" wizard would just do this automatically instead of manually.

PS: Alternately this sort of "forking" may be done with Git instead (with use of tags or branch names)

ollelogdahl, 2020-10-11:
I'd like to contribute to the project. looking into this!

ubruhin, 2020-10-13:
I think a "Save As..." would be the most intuitive variant (as it is well known from other programs), but probably also the most complicated one. We'd have to be very careful to really update the filepaths of all objects to avoid accidentally accessing the original project files after saving a projects to a new location.

Btw, I suggest to also assign a new UUID to the project (see class `ProjectMetadata`). The UUID has no clear technical functionality, but might be useful some day to distinguish between different projects.

> I'd like to contribute to the project. looking into this!

That would be great! :)

ollelogdahl, 2020-10-14:
Should this button be an option next to 'Save' option on both the Board Editor and the Schematic editor?

miniwark, 2020-10-24:
> Should this button be an option next to 'Save' option on both the Board Editor and the Schematic editor?

I think the best place would be in the main Control Panel instead.