LibrePCB/LibrePCB #762: Board editor: Assert is raised when starting trace at unconnected pad

ubruhin, 2020-08-31:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.2.0-unstable
Git Revision:     2040eac1afbc0cf1f135884d9c05bf62b156a21c
Build Date:       2020-08-31T20:59:09
Linking type:     static
Unbundled libs:   -
Qt Version:       5.5.1 (built against 5.5.1)
CPU Architecture: x86_64
Operating System: Ubuntu 16.04.7 LTS
Platform Plugin:  xcb
```

##### SUMMARY

When starting to draw a trace at a footprint pad which has no netsignal, an assert is raised (resp. crashes in release mode).

##### STEPS TO REPRODUCE

- Add a device to a board containing pads which are not connected
- Start the "draw trace" tool
- Click on the unconnected pad

##### EXPECTED RESULTS

The trace should start at the pad.

##### ACTUAL RESULTS

The following assert is raised:

https://github.com/LibrePCB/LibrePCB/blob/1f2c0d411a165ae42772b49549dbef78a3c91794/libs/librepcb/projecteditor/boardeditor/fsm/bes_drawtrace.cpp#L566

/cc @5n8ke

5n8ke, 2020-08-31:
I'll take a look

5n8ke, 2020-08-31:
Done. There was an exception missing to check for unconnected pads. Every other item used for drawing traces shouldn't/can't exist without a netsignal.

5n8ke, 2020-09-01:
@ubruhin On closer inspection this morning, I'm actually not sure how this bug can be reproduced with the current master version since findPad() does only return pads that are connected to a NetSignal.
This does result in the already present error "Nothing to connect here".

Could you supply me with a save that has this issue?

ubruhin, 2020-09-01:
Whooops, it seems I mixed up things :scream: Probably I checked out #753 locally when I experienced this, and thought it also happens on `master` since it also crashed when clicking *on* a pad, not just when clicking *beside* it :sob: I didn't realize that clicking *on* the pad still triggers `findAnchorNextTo()`. So the fix in #753 seems to be all we need.

Sorry for the confusion! :slightly_smiling_face:

5n8ke, 2020-09-01:
No problem. It's good that the error was caught early. And it may as well have happened on master if not the findPad routine was behaving differently from the other functions

PR #764 (2020-09-02): BES_DrawTrace: Get rid of member variable mCurrentNetSignal
Fix crash when trying to connect to a pad that is not part of a signal.

Fixes #762
closed 2020-09-02 by ubruhin