LibrePCB/LibrePCB #766: Upgrade minimal supported Qt version

5n8ke, 2020-09-03:
To keep up with the times and use the newly available features of Qt, especially with the release of Qt6 on the horizon, a gradual upgrade of the minimal supported Qt version is necessary.
The current minimal supported version is Qt 5.2 which was released at the end of 2013.

The minimal version is determined by the currently oldest LTS version of Debian and Ubuntu. Both Debian and Ubuntu offer extended security maintenance modes, that extends the standard lifetime by 3 to 5 years. Only the standard end of life is considered and not the extended security support.

Below follows an overview of the standard end of life for the currently active Debian and Ubuntu versions along with their supported Qt version. Releases that reached their end of life are marked in italics.

| Date | Release| Supported Qt version |
|---|---|:---:|
| _2019/04_ | _Ubuntu 14.04_ | _5.2.1_ |
| _2020/06_ | _Debian 8_ | _5.3.2_ |
| 2021/04 | Ubuntu 16.04 | 5.5.1 |
| 2022 | Debian 9 | 5.7.1 |
| 2023/04 | Ubuntu 18.04 | 5.9.5 |
| 2024 ? | Debian 10 | 5.11.3 (still updating) |
| 2025/04 | Ubuntu 20.04 | 5.12 (still updating) |

### Sources
- https://ubuntu.com/about/release-cycle
- https://packages.ubuntu.com/search?keywords=qt5-qmake&searchon=names&suite=all&section=all
- https://wiki.debian.org/DebianReleases
- https://packages.debian.org/search?keywords=qt5-default&searchon=names&suite=all&section=all

ubruhin, 2020-09-05:
Thanks for the nice overview! :+1:

So now I'd agree to drop support for Qt versions older than 5.5 since both Ubuntu 14.04 and Debian 8 are EOL.

Tasks to do the upgrade:
- Update README.md
- Update Doxygen page "Code Style Guide"
- Remove all the `#if / #endif` which were added to support Qt<5.5

5n8ke, 2020-09-05:
The test and build pipelines also need an update and the 14.04 image can be dropped.

ubruhin, 2021-04-26:
FYI, I opened PR #885 to drop support for Qt < 5.5.

ubruhin, 2024-05-06:
Closing this as Qt 5.12 is now the minimum requirement and soon we will drop Qt5 support entirely.

closed 2024-05-06 by ubruhin