LibrePCB/LibrePCB #778: Headers of fontobene-qt5 are deployed with "make install"

ubruhin, 2020-10-04:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Manjaro Linux
Platform Plugin:  xcb
```

##### SUMMARY

When running `make install`, the header files of fontobene-qt5 are installed to `/include`, although this does not make any sense (we want to install *LibrePCB*, *not* fontobene-qt5!).

##### STEPS TO REPRODUCE

Run `make install` and check the installed files.

##### EXPECTED RESULTS

No directory `/include` should be installed at all.

##### ACTUAL RESULTS

A directory `/include` containing the fontobene-qt5 headers gets installed.

##### ADDITIONAL INFORMATION

Because of this issue, even our binary releases contain that useless directory. For example [librepcb-0.1.5-linux-x86_64.tar.gz](https://download.librepcb.org/releases/0.1.5/librepcb-0.1.5-linux-x86_64.tar.gz).

dbrgn, 2020-10-04:
Oh, the perils of qmake :slightly_smiling_face: 

Can we somehow exclude the entire `fontobene-qt5` directory from being installed recursively?

ubruhin, 2020-10-04:
If I remember correctly, I already investigated this issue some time ago and didn't find a solution. Then I forgot this until today where I saw the `/include` directory again :see_no_evil:

ubruhin, 2021-09-14:
Should be fixed with #798.

closed 2021-09-14 by ubruhin