LibrePCB/LibrePCB #782: Library structure not easy to be used in a real industry

mesmariusz, 2020-10-08:
This is difficult to build an individual "unchangable" (released) component, e.g. 
R=10k_T=5%_P=0.25W_Pk=R-THT-0207_Raster=7.62mm

including all production information (manufacturer, partnumber, manufacturer datasheet link, PCBA (internat) datasheet, PCBA THT bending type number) and so one. 

For the industry the "component" (component released inside the industry system) is something completly unchangable (no any other raster, no any other bending type, no any other notheing is allowed)

Currently (current LibrePCB library concept) the only one way is to put such kind of information into "component" element:

https://i.imgur.com/sQB7GW5.png 
https://i.imgur.com/sAJ0nPA.png

But this is not consistent into general library idea.

Just look into official one library, where the "components" are just an "type of components" instead of just one lingle, released @system component:

https://i.imgur.com/f5AZIRq.png

And if you try to think about ad "device" as fully defined component, this is impossible to introduce necessary data (e.g. manufacturer, partnumber, manufacturer datasheet link, PCBA (internat) datasheet, PCBA THT bending type number) and so one).

https://i.imgur.com/nBtngqN.png

##### SUMMARY
Please make the library concept usable for real PCB industry (or just for proffessional hobbist usage). Let's avoid to create next "library monster", unusable and living it's own live). Let's follow the real industry way. 

##### SOLUTION
Please modify in a reasonable way or introduce into library the group for real, released, unchangable, fully defined components (e.g. assembled from all other groups, but finally unconfigurable, just possible to use and notheing more).

ouabache, 2020-10-09:
LibrePCB's library is fine. You need to understand how it is used.

Modern design is done by teams where each designer does a task and then passes the design onto the next one. You do not want one library with everything in it, you want each designer to only see what is relevant to their task.

Circuit designer selects a resistor and specifies a value,tolerance and power rating. The design is then passed to the board designer who selects the package and then onto the procurement engineer who selects the actual part.

The circuit designer uses COMPONENTS and may not even care about package options or who makes the part. They could select a DEVICE if they know what is is but it is fine if they don't.

The board engineer takes the schematic and assigns each COMPONENT to a DEVICE to select the PACKAGE. They can even override the Circuit designers choice. 

The procurement engineer takes the list of DEVICES and maps them to manufacture part numbers. LibrePCB does not contain this information because it is only concerned with the circuit designer and board engineer and they don't care who makes the part.

We need a new library for manufacturer data that maps back to the current DEVICE libraries. We should not add manufacturer data into the current libraries.

mesmariusz, 2020-10-09:
Up to the end of this sentence I (almost) fully agree:

"Circuit designer selects a resistor and specifies a value, tolerance and power rating. The design is then passed to the board designer who selects the package and then onto the procurement engineer who selects the actual part".

But! He isn't allowed to do it by himself!

I don't know too much about the place of your work, and about your industry experiences, but follow our development structure, if PCB designer would like to change somehow a component (e.g it's package or type), then HAVE TO ASK the hardware engeneer to change the circuid diargam, e.g. to change some choke from THT to SMT on the circuit diagram. The hardware engeneer must do it according his deep knowledge and huge experience (it's his design and his responsibility). Layut engeneer is not allowed to change any hardware on the circuit diagram! Layout engeneer is just a layout engeneer! And notheing more!

The hartware engeneer must be able to use fully defined components. Layout engeneer just place it on PCB and connect (notheing more). If he would like to change anything then he needs to contact hardware engeneer and ask him to change the circuit diagram (circuit diagram contain fully defined components). Even THT bending type for legs for prepared compoents (e.g. high power axial resistors which needs to be placed above the PCB surface (to not touch it) for better cooling itself) must be defined @ component used @ circuit diagram. Every single components has an specific internal number. Two exactly the same resistors but which different tye of pin bending (prepared THT components) that's are two different components (two different internal numbers @ BOM) two different components in a library!

That's why I noticed current library concept as difficult for the real industry. But probably easy to improove...

At the moment, the only way to meet mentioned requirement is to build as many components as it's dedicated devices:

[image]

because every single device is build from "component" and "package":

[image]

For mentioned industry requirement it would be much easier if such fields:

[image]

could be specified on the "final" device, instead of @ component. 

I hope you understand my "industry thinking".

Current concept is great and usefull especially for hobbyists which do not need to follow fully defined partnumbers. 

This is really great, but we need (we should) also take care about professional / industry users. And I think it could be easy to add a feature which could make it possible to create fully definec components (paralelly for current solution). 

Let's think about it. Let's try to introduce. Let's improve :-) 

Best regards
Mariusz

mesmariusz, 2020-10-09:
Ps. Please note it's easy to introduce. We just need an additional group e.g. "fully defined components" inside we could keep connections between components and devices and introduce some additional information like manufacturer partnumber, internal part number, bending drawing number, datasheet and a lot of other very specific details. 

It's possibile. And seems to be easy :-)

mesmariusz, 2020-10-10:
PS. It might be even the first software which hint the fully defined components based on layout designer chooses (and report to hardware engineer missing components (based on layout designer chooses) to create and release it.

ouabache, 2020-10-10:
So for what changes does the layout engineer have to go back and ask the circuit designer to modify the schematic?

Package Selection? Pin swapping? Gate swapping? Reference designator reordering? 

What if the circuit designer creates a reference design that goes into several products. Each one will be different and the designer does not want tp have tp customize the schematic for each one.

You only put what is essential in the schematic and let the layout engineer do their own customizations

mesmariusz, 2020-10-10:
Well, just start working asvlayout designer for a real company, e.g. Bosh, Electrolux, Whirlpool etc. And then let's back to this discussion again. Otherwise probably you will not be able to understand what I'm taking about. 

Because suche kind of thinking LibreBCB will stusck like other open softwares with potentially "huge potential".

EdizonTN, 2020-10-10:
Salute to all,
please, we take a pause for a moment and try small thinking.

FACTS:
Fact 1: The discussants use a different scenario to design the circuits and then the PCB
Fact 2: None of the EDA tools will cover all kinds of scenarios (I don't know one)
Fact 3: The designer should not adapt to the application - on the contrary, he chooses the application that suits him. It also applies to design departments in companies.
Fact 4: Citation from the key feature of the LibrePCB is "Very powerful library design with some innovative concepts"

Questions:
Question 1: Do you want to expand the use of LibrePCB to use more designers, or is it just a hobby project??
- the author should answer?
Question 2: Does LibrePCB have the potential to expand among designers?
- I think so (this is the view of the user, not the programmer), but be realistic - this is a long way.

Then I don't understand why adding / editing a few attributes in the library is such a problem, especially if it improves the application, taking into account fact 1 to 4!

I think it would be a really pity to lock down the possibility of using LibrePCB at this stage of development, just for a few missing attributes.

There are several professional PCB designers between LibrePCB users, so let's appreciate it and try to take advantage and hear theirs.

My suggestion for discussion (maybe it would be appropriate to create a separate thread for this if you are interested to discuss):
It would be good if each designer described his own scenario (or design flow). But please, let's not discuss the scenarios, each have their own reason why work just that (and has their / client's / corporate reasons for it). This discussion is useless.
We should discuss methods of work with LibrePCB, to meet the scenarios in the easiest way possible.
LibrePCB programmers should focus on how to modify LibrePCB to be able to implement as many of these scenarios as possible. In the easiest way of course from a user point of view.

And finally: These scenarios should be in the documentation for the program (in graphical form)!
Of course, the documentation should also include design scenarios that cannot be implemented in LibrePCB (it's fair).
This would clearly demonstrate the capabilities of LibrePCB and would be a key criterion in creating user libraries (and not only libs).

After all - good libbraries are the main basis for good design!

Ed.

mesmariusz, 2020-10-11:
So I think nie is (maybe) the right timer and (maybe) the last chance (early development phase) to introduce as additional one mentioned corporate scenario for workflow. 

It's easy because to do that we need just a one additional column (group) of components (called fully defined components). 

The best if the hardware engineer who is drawing the scircuit diagram choose the component

ubruhin, 2020-10-11:
I totally agree the library concept is not 100% finished yet, which is currently one of the weaknesses of LibrePCB especially for professional use. Of course my goal is to extend the library concept to fix this issue. The library should be able to contain attributes like resistor values, MPNs, datasheet URLs and so on, while still keeping the ability to draw schematics with non-concrete components.

I hope we can define and implement such a concept for LibrePCB 0.2.0. Probably I'll start a discussion about it in our discussion forum first as soon as I start with this task.

BartLH, 2020-11-20:
> So for what changes does the layout engineer have to go back and ask the circuit designer to modify the schematic?
> 
> Package Selection? Pin swapping? Gate swapping? Reference designator reordering?

Only the first item is relevant for a discussion about the library design, and @mesmariusz is completely right there. Layout engineers do *not* do package selection. They might suggest an improvement to the design engineer based on their experience, but the design engineer is the one who decides if it gets changed or selects it. The layout engineer does not have sufficient knowledge about the amount of power that needs to be dissipated, the influence of lead length parasitics, etc.

I have no opinion on how (and whether) the library should be (re)restructured to accommodate that (or at least, not yet), but it seems to me that the workflow you have in mind is not an accurate representation of professional use.

ouabache, 2020-11-21:
Layout engineers do not do package selection. They might suggest an improvement to the design engineer based on their experience, but the design engineer is the one who decides if it gets changed or selects it. The layout engineer does not have sufficient knowledge about the amount of power that needs to be dissipated, the influence of lead length parasitics, etc.

==================
Circuit designers do not do package selection. They pick a part with values and tolerances and spice the circuit to ensure that it works. They may lay out and build a reference board to fully verify their design but they will pick packages that are easy for them to assemble and not what will eventually be used in production. They then put their design out on github.com.

Dozens of projects will download and incorporate the circuit on their boards and they will all have different package requirements. None of which will be sent back to the original designer. You could never support all the various packages that will be used in your design.

Modern design is a two step process. One engineer picks the component and a second engineer picks the package. You may be doing both of those roles but we still need to be able to separate  them.

Sometimes it is a three step process. If you call out a thru hole resistor with 10% tolerance and I have a reel of them on the pick and place machine with 5% tolerance then guess what is going on the board? The circuit designer calls out the design minimums and production can use any part that meets or exceeds those requirements. That's what you reorder.

doom-fr, 2021-02-10:
Firstly, I need to say that I am the only hardware guy in my company, so I am both design engineer & layout engineer.
So I do not really know the real world. Despite this, and regarding my experience about industrialisation, I agree with @mesmariusz.

What I can say is that same component in different package (same starting reference, sometimes in same datasheet) do not have same characteristics. And so, it most cases, it cares. But sometimes not.

For example some components, reference is really important : diode, mosfet, ...
For other not really : resistors 
 - often I need a 1005 resistor of 10k, but I do not care about the technology or the manufacturer, and the worst precision would be ok
 - but sometime, it is important : for RF resistors for example

What I can say is that the **BOM should be directly generated from schematic** without manual changes and without the need to add extra attributes to device (except special needs like bending of power resistors :smile:  )
And with this BOM, your supplier need to be able to tell you a price and to purchase this for you.

Regarding this, I think there's a problem in lib structure as **devices need to have attributes**.

Another problem is that we could not really add a device for each resistor reference that exists. All the more that a lot will be exactly the same except the value.

So **we need to create the concept of _series_**. In series of devices, there will be common attributes (package, power, ...) and a listing of device with there additional attributes (value, ...)
When selecting the device, LibrePCB will propose available devices of loaded series regarding the value already entered.

Maybe this should be extended to a tree structure.

**PS :** I want to thanks here all the LibrePCB team to share us this amazing software, even if it is not fully perfect now. It is really usable and I already experienced board industrialisation without major problems. :clap: :clap: :clap:

doom-fr, 2021-02-10:
In addition, in part reference, it might be also precision of the shipping (reel, tray, sometimes, reel of 4k and reel of 8k).
So exactly same device with same package, just different way to store it.

And be careful, for example, Microchip put a "T" to say tape and reel (nothing is tray), but before the part of the reference that define the package or the temperature range. I found really strange to do that.

i9e1, 2022-04-19:
As Ed said its useless to discuss workflows, but I think the two extremes here have different levels of communication between the circuit and pcb designer (who is allowed to choose e.g. the package). 
Would a feature like a lock / semaphore for devices in the schematic help here? I understood 
@mesmariusz workflow like that it asking the circiut designer for everything is kind of company enforced "locking".
Imagine the circuit designer chose a specific package because of e.g. power dissipation. Then she could set a lock (maybe with a comment) which would block a device change in pcb design. The pcb-designer than would need to either deal with it or ask. So some team communication is "supported" by EDA.
On the other hand, if one does circiut and pcb it may be practical to have kind of a "harder" comment (not just some text). 

Does this make any sense for you? Or do you think no one would really use such a lock? 
Or is it more like an extra documenting feature unconnected to libs?

closed 2023-07-18 by ubruhin