LibrePCB/LibrePCB #785: Build fails whenever attempting to unbundle a library

lgbaldoni, 2020-10-13:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.5
Git Revision: stable
Qt Version: 5.15.1
Operating System: openSUSE Tumbleweed
```

##### SUMMARY
Whenever the UNBUNDLE variable is used, the build fails with a linking error.

##### STEPS TO REPRODUCE
Run qmake passing e.g. UNBUNDLE="fontobene-qt5"

##### EXPECTED RESULTS
That system-installed libraries were used and correctly linked to.

##### ACTUAL RESULTS
The build fails during the linking phase, apparently always in the same way, regardless of the library one tries to unbundle.

##### ADDITIONAL INFORMATION
See [log snippet](https://github.com/LibrePCB/LibrePCB/files/5370820/2359908e.txt).

dbrgn, 2020-10-16:
Should be fixed when:

- https://github.com/LibrePCB/quazip/pull/2 is merged
- The quazip submodule is updated to include the fix above
- #788 is merged

To test:

```
mkdir build-dynamic-quazip && cd build-dynamic-quazip
qmake -r .. CONFIG+=ccache UNBUNDLE+=fontobene-qt5
make -j8
```

ubruhin, 2021-09-14:
I think this should now be fixed with #798. In case the problem still exists with the current master branch or the upcoming 0.1.6 release, don't hesitate to reopen this issue.

closed 2021-09-14 by ubruhin