LibrePCB/LibrePCB #792: Unbundling libraries lead to messy project files and doesn't work reliable

ubruhin, 2020-10-17:
##### SUMMARY

In the past, we always linked statically to our dependencies bundled as submodules. Then we started to optionally allow dynamic linking and unbundling libraries: #698, #699, #742, #779, https://github.com/fontobene/fontobene-qt5/pull/2, https://github.com/fontobene/fontobene-qt5/pull/5, https://github.com/LibrePCB/docker-librepcb-dev/pull/4, https://github.com/LibrePCB/docker-librepcb-dev/pull/8

Due to these changes, many new problems popped up and required bugfixes or workarounds: #729, #778, #788, #789, #790, #791, https://github.com/fontobene/fontobene-qt5/pull/8, https://github.com/LibrePCB/quazip/pull/2, https://github.com/LibrePCB/quazip/pull/3, https://github.com/LibrePCB/docker-librepcb-dev/pull/5

Slowly I really struggle with all the dynamic linking and unbundling stuff. Before starting with these things, everything worked fine, it was very easy to build, on all platforms. The only problem was that packagers do not like bundled libraries. Now the project files get more and more complicated (or even messy!) to fix or work around problems introduced by the unbundling stuff. And most of these issues can't even be detected on CI as this would require building with dozens of different configurations. So for me this looks like a never ending story, wasting a lot of time and nerves.

Therefore I think this can't be the way to go. How do other projects handle dependencies to other libraries? Would CMake and/or a package manager like Conan be able to handle all use cases (static linking, dynamic linking, bundled libraries, system libraries) in a reliable way? What other options do we have to get a more reliable build system?

Please don't get me wrong @dbrgn - I really appreciate your work to help packagers adding LibrePCB to package managers. But now I really wonder if this is the way to go...

dbrgn, 2020-10-17:
I think most projects just link dynamically. A lot of the complexity right now doesn't come from "bundled vs system", but from "dynamic vs static".

I think if we'd create dynamically linked builds by default, that would make things easier already. Switching to CMake should also make things easier, because many libraries provide CMake scripts, so we can just bundle them without having to adjust build scripts (we could even use a "download external libraries" script instead of submodules, which might make things easier). CMake also has tools for library discovery, so we don't have to hardcode paths.

Not bundling libs at all would also be an option, but it would make it a bit more complex for contributors, especially on Windows.

Here's an overview of the libs we're using:

| Lib | CMake? | Available on current Ubuntu LTS? | Available on current Arch? | Available on Fedora? |
|-|-|-|-|-|
|quazip|yes|yes|yes|yes|
|fontobene-qt5|no (easy to fix)|no|yes|no|
|polyclipping|yes|yes|yes|yes|
|hoedown|no (but makefile)|no|no|no|
|sexpresso|no ([pr available](https://github.com/BitPuffin/sexpresso/pull/5)|no|no|no|
|muparser|yes|yes|yes|yes|
|type_safe|yes|no|no|no|
|delaunay-triangulation|no (meson)|no|no|no|
|optional|yes|no|no|no|
|googletest|yes|yes|no|yes|
|parseagle|no (easy to fix)|no|no|no|

As you can see, almost all libs use CMake as the build system. We'd definitely profit from being able to include them more easily, without messing with custom qmake scripts.

CMake seems to be quite well supported by qtcreator, not as well as qmake, but it'll get better. My suggestion would be to switch to CMake, with dynamic builds by default. For easier building, libs should either be vendored as submodules, or downloaded into a local directory using a "fetch dependencies" script (maybe that can even be scripted using CMake).

Regarding the minimum version of CMake required (this is important in order for modern features to be useable), I'd go with at least 3.0 (Ubuntu 16.04 ships 3.5, Fedora 30 ships 3.14, Debian 9 ships 3.7).

ubruhin, 2020-10-17:
Btw, maybe we should also rethink about our [software architecture](https://developers.librepcb.org/de/d34/doc_architecture_overview.html) to try making it simpler. For example, maybe we could merge the four libraries `librepcb/common`, `librepcb/library`, `librepcb/project` and `librepcb/workspace` into a single library? Somehow I like to have separate libraries since it helps to avoid architecture violations, but merging them together might make the whole build system simpler.

dbrgn, 2020-10-17:
CC @lgbaldoni

ubruhin, 2020-10-17:
> I think most projects just link dynamically. A lot of the complexity right now doesn't come from "bundled vs system", but from "dynamic vs static".

True! I wonder if it would be possible to *always* link dynamically (i.e. no longer support static linking). My thoughts about that:
- Should work fine on Windows as long as the DLLs are located next to the executable.
- On binary deployments for linux, the *.so files probably have to be located in the `$PREFIX/lib` directory, otherwise they are not loaded. Unfortunately this is not consistent with the DLL location on Windows. When deploying *.so files next to the executable, we'll need some workaround using `LD_LIBRARY_PATH`, `RPATH` or something like that.
- For official Linux distribution packages (e.g. *.deb), is it allowed to deploy 3rd party libraries within `$PREFIX/lib`? Probably not, since for example when the `librepcb` package installs the file `/usr/lib/sexpresso.so`, and other applications do the same, it would lead to a conflict. How to handle this situation? Prefixing the library (e.g. installing to `/usr/lib/librepcb-sexpresso.so`) sounds quite ugly. So, where to install such a bundled, shared library?

lgbaldoni, 2020-10-17:
> For official Linux distribution packages (e.g. *.deb), is it allowed to deploy 3rd party libraries within $PREFIX/lib? Probably not, since for example when the librepcb package installs the file [...]

openSUSE uses /usr/lib or /usr/lib64 depending on architecture (cf. https://github.com/LibrePCB/LibrePCB/pull/793, https://github.com/LibrePCB/quazip/pull/4). Also plain .so is used only for linking and it goes in a separate -devel package or is deleted outright.
However _unversioned_ shared libraries can be installed in custom paths but of course they'd have to be treated like plugins.

ubruhin, 2020-10-24:
OK, I researched a bit about this and found that it seems to be common practice to put application-specific shared libraries into a subdirectory of `$PREFIX/lib` (example: `/usr/lib/librepcb/sexpresso.so`). Many applications are doing this, including Firefox, Chromium, VLC, ... I suspect these *.so files are usually loaded like plugins at runtime, i.e. not linked against them.

So if we link LibrePCB dynamically, we could install the libraries into `$PREFIX/lib/librepcb/`. But then the question is still whether to actually link against them, or to load them at runtime. Somehow I prefer to explicitly link against them (might avoid issues for example with AppImage since the dependency to plugins can't be detected automatically), but then we'd need some executable wrapper script which extends `LD_LIBRARY_PATH` :thinking: 

Btw, some applications even put their executable into the `/usr/lib/<application>/` folder, although I'm not sure for what reason. For example Firefox under Arch Linux:
- `/usr/bin/firefox` is a shell script containing just `exec /usr/lib/firefox/firefox "$@"`
- `/usr/lib/firefox` seems to be the actual executable
- `/usr/lib/firefox-bin` is a symlink to `/usr/bin/firefox` (no idea what's its purpose)

Some links about this topic:
- [What's the accepted method for deploying a linux application that relies on shared libraries?](https://stackoverflow.com/questions/7099712/whats-the-accepted-method-for-deploying-a-linux-application-that-relies-on-shar)
- [Arch package contents of Firefox](https://www.archlinux.org/packages/extra/x86_64/firefox/files/)
- [Linux Directory Hierarchy](https://tldp.org/HOWTO/HighQuality-Apps-HOWTO/fhs.html)

dbrgn, 2020-10-25:
Btw, with qmake I think this is more tricky, but with cmake it should be fairly easy to statically link the librepcb libs and vendored libs, while dynamically linking against system libs.

FedericoCeratto, 2020-12-26:
For a more detailed directory hier: https://manpages.debian.org/jessie/manpages/hier.7.en.html

dbrgn, 2020-12-26:
With CMake (see #798) it should become quite flexible on how to build and link LibrePCB. Currently still fighting over a few linker errors, but it's coming along well.

ktrace, 2021-02-09:
> OK, I researched a bit about this and found that it seems to be common practice to put application-specific shared libraries into a subdirectory of `$PREFIX/lib` (example: `/usr/lib/librepcb/sexpresso.so`).

> So if we link LibrePCB dynamically, we could install the libraries into `$PREFIX/lib/librepcb/`.

I think `$PREFIX/lib` incorrect solution. So I already stuck on gentoo with "multilib-strict check failed!", 'cause we must install libs in /usr/lib64. Seems more correct `target.path = $${QT_INSTALL_LIBS}` in *.pro.

Must I make Issue or/and PR?

dbrgn, 2021-02-09:
@ktrace if you want, check out #798 and test how well it works for you. On Linux, it should already be mostly working. The main open issues are packaging on macOS and Windows. I'll tackle them as soon as I find time.

ktrace, 2021-02-09:
I like the idea of switching to CMake. As long as you give up qmake, issue and PR are unnecessary. Thank you, I'll try CMake build, but on current release I'll just patch .pro.

dbrgn, 2021-06-07:
The `cmake` branch is now ready for testing! More details can be found here: https://github.com/LibrePCB/LibrePCB/pull/798#issuecomment-856229198

CC @lgbaldoni @FedericoCeratto @ktrace @FFY00

lgbaldoni, 2021-06-08:
@dbrgn muparser 2.3.2 doesn't come with .cmake files, unless 2.3.3 is required we need a pkgconfig fallback.

lgbaldoni, 2021-06-08:
Another couple of minor things:

this is installed even with `-DUNBUNDLE_FONTOBENE_QT5=ON`
```
/usr/share/librepcb/fontobene/.gitattributes
/usr/share/librepcb/fontobene/README.md
```
this could be an artifact of the git snapshot I'm using;
```
/usr/share/librepcb/fonts/.gitattributes
```
and couldn't we use the Noto system font if available?
```
/usr/share/librepcb/fonts/NotoSans-Regular.ttf
/usr/share/librepcb/fonts/NotoSansMono-Regular.ttf
```

dbrgn, 2021-08-29:
> @dbrgn muparser 2.3.2 doesn't come with .cmake files, unless 2.3.3 is required we need a pkgconfig fallback.

Fixed.

> this is installed even with -DUNBUNDLE_FONTOBENE_QT5=ON

These are the fonts shipped with LibrePCB (they do not belong to the fontobene package itself). Since they are included with a submodule, these git meta-files exist and are copied along. If you download an official build tarball, those files will not be there.

Nevertheless, I found a way to exclude git files when bundling shared files:

```diff
 foreach(d ${SHARE_DIRS})
     if(IS_DIRECTORY "${CMAKE_CURRENT_SOURCE_DIR}/${d}")
-        install(DIRECTORY ${d} DESTINATION ${CMAKE_INSTALL_DATAROOTDIR})
+        install(
+            DIRECTORY ${d}
+            DESTINATION ${CMAKE_INSTALL_DATAROOTDIR}
+            REGEX "/\.(git|gitattributes|gitignore)$" EXCLUDE
+        )
     endif()
 endforeach()
```

> and couldn't we use the Noto system font if available?

I'm not sure what the implications are. If this is important to you, please open a new issue to discuss it.

closed 2021-09-14 by ubruhin