LibrePCB/LibrePCB #799: Schematic Editor: Search by text in component dialog crashes kicad

baryluk, 2020-10-27:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 5.1.7 from Debian testing
wxWidget: 3.05 / 3.0.5.1
glib 2.66.1-2
gtk3 3.24.23-2
Linux amd64
```

##### SUMMARY
If I place a relay with sub components, and then want to add another component, lets say a connector, I go to the component dialog again, and when I press "c" key in the search input, the program crashes immedietly.

##### STEPS TO REPRODUCE

* Create new schematic.
* Go to a component tool, click on a sheet, to place a component. Components will be parsed and component dialog will show up.
* In a component dialog, search for `FRT5_separated` component.
* Put 3 units (Unit A, Unit B, Unity C) of a FRT5 one by one on the schematic. Placing just Unit A, is also fine to reproduce it.
* Go again into component dialog to place another / different component.
* Press "c" key in the search input field.
* Crash.

Stack trace:

```
Thread 1 "kicad" received signal SIGSEGV, Segmentation fault.
0x00007ffff7e700fe in wxDataViewCtrlInternal::BuildBranch (this=0x55555efc0b90, node=0x0) at ../src/gtk/dataview.cpp:3489
3489	../src/gtk/dataview.cpp: No such file or directory.
(gdb) bt full
#0  0x00007ffff7e700fe in wxDataViewCtrlInternal::BuildBranch(wxGtkTreeModelNode*) (this=0x55555efc0b90, node=0x0) at ../src/gtk/dataview.cpp:3489
#1  0x00007ffff7e7199e in wxDataViewCtrlInternal::iter_nth_child(_GtkTreeIter*, _GtkTreeIter*, int) (n=0, parent=<optimized out>, iter=0x7fffffffa1d0, this=0x55555efc0b90) at ../src/gtk/dataview.cpp:4094
        id = <optimized out>
        item = {<wxItemId<void*>> = {m_pItem = 0x55555b8f35d0}, <No data fields>}
        parent_node = 0x0
#2  wxDataViewCtrlInternal::iter_nth_child(_GtkTreeIter*, _GtkTreeIter*, int) (this=0x55555efc0b90, iter=0x7fffffffa1d0, parent=<optimized out>, n=<optimized out>) at ../src/gtk/dataview.cpp:4060
#3  0x00007ffff688c5eb in gtk_tree_row_reference_unref_path_helper (current_depth=2, depth=3, parent_iter=0x7fffffffa1b0, model=0x55556694dd30, path=0x555556335210) at ../../../../gtk/gtktreemodel.c:2273
        iter = {stamp = 0, user_data = 0x0, user_data2 = 0x0, user_data3 = 0x0}
        iter = {stamp = -715386592, user_data = 0x55555b8f35d0, user_data2 = 0x0, user_data3 = 0x0}
        iter = {stamp = -715386592, user_data = 0x55555b8f3780, user_data2 = 0x0, user_data3 = 0x0}
#4  gtk_tree_row_reference_unref_path_helper (current_depth=1, depth=3, parent_iter=0x7fffffffa190, model=0x55556694dd30, path=0x555556335210) at ../../../../gtk/gtktreemodel.c:2274
        iter = {stamp = -715386592, user_data = 0x55555b8f35d0, user_data2 = 0x0, user_data3 = 0x0}
        iter = {stamp = -715386592, user_data = 0x55555b8f3780, user_data2 = 0x0, user_data3 = 0x0}
#5  gtk_tree_row_reference_unref_path (path=0x555556335210, model=0x55556694dd30, depth=3) at ../../../../gtk/gtktreemodel.c:2289
        iter = {stamp = -715386592, user_data = 0x55555b8f3780, user_data2 = 0x0, user_data3 = 0x0}
#6  0x00007ffff688c914 in gtk_tree_row_reference_unref_path (depth=<optimized out>, model=<optimized out>, path=<optimized out>) at ../../../../gtk/gtktreemodel.c:2285
        refs = 0x5555662636c0
        __func__ = "gtk_tree_row_reference_free"
        __func__ = "gtk_tree_row_reference_free"
#7  gtk_tree_row_reference_free (reference=0x555566268f50) at ../../../../gtk/gtktreemodel.c:2522
        refs = 0x5555662636c0
        __func__ = "gtk_tree_row_reference_free"
        __func__ = "gtk_tree_row_reference_free"
#8  gtk_tree_row_reference_free (reference=0x555566268f50) at ../../../../gtk/gtktreemodel.c:2496
        __func__ = "gtk_tree_row_reference_free"
#9  0x00007ffff68a8aff in _gtk_tree_view_set_anchor_path (tree_view=0x55555609a7f0 [GtkTreeView], anchor_path=anchor_path@entry=0x0) at ../../../../gtk/gtktreeview.c:10238
#10 0x00007ffff6898e20 in gtk_tree_selection_real_unselect_all (selection=selection@entry=0x5555561dfab0 [GtkTreeSelection]) at ../../../../gtk/gtktreeselection.c:1230
        tree = 0x5555662b2150
        node = 0x555556089060
        anchor_path = 0x555556307400
        priv = 0x5555561dfa80
        tuple = <optimized out>
#11 0x00007ffff6898e8b in gtk_tree_selection_unselect_all (selection=0x5555561dfab0 [GtkTreeSelection]) at ../../../../gtk/gtktreeselection.c:1281
        priv = 0x5555561dfa80
        __func__ = "gtk_tree_selection_unselect_all"
#12 0x00007ffff7e7430d in wxDataViewCtrl::UnselectAll() (this=0x555566267200) at ../src/gtk/dataview.cpp:5054
        selection = <optimized out>
#13 0x00007fffd8d060f1 in LIB_TREE_MODEL_ADAPTER::UpdateSearchString(wxString const&) (this=0x555556d6b7b0, aSearch=...) at ./common/lib_tree_model_adapter.cpp:163
        updateLock = {m_win = 0x555566267200}
        tokenizer = {<wxObject> = {_vptr.wxObject = 0x55555572aa38 <vtable for wxStringTokenizer+16>, static ms_classInfo = {m_className = 0x7ffff74bb0d0 L"wxObject", m_objectSize = 16, m_objectConstructor = 0x0, m_baseInfo1 = 0x0, m_baseInfo2 = 0x0, static sm_first = 0x0, m_next = 0x7ffff7547f40 <wxMemoryOutputStream::ms_classInfo>, static sm_classTable = 0x555555745710}, m_refData = 0x0}, m_hasMoreTokens = wxStringTokenizer::MoreTokens_No, m_string = {static npos = 18446744073709551615, m_impl = L"c", m_convertedToChar = {m_str = 0x0, m_len = 2}}, m_stringEnd = {m_cur = 0 L'\000'}, m_delims = {<wxCharTypeBuffer<wchar_t>> = {<wxScopedCharTypeBuffer<wchar_t>> = {m_data = 0x555556abe2b0}, <No data fields>}, <No data fields>}, m_delimsLen = 4, m_pos = {m_cur = 0 L'\000'}, m_mode = wxTOKEN_STRTOK, m_lastDelim = 0 L'\000'}
        bestMatch = <optimized out>
[... 414 lines cut ...]
#98 0x00007ffff6157e48 in g_main_context_iterate (context=0x555555807420, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../glib/gmain.c:4092
        max_priority = 300
        timeout = 0
        some_ready = 1
        nfds = <optimized out>
        allocated_nfds = <optimized out>
        fds = 0x555556498af0
#99 0x00007ffff615813b in g_main_loop_run (loop=loop@entry=0x55555d05ac20) at ../../../glib/gmain.c:4290
        __func__ = "g_main_loop_run"
#100 0x00007ffff67797a5 in gtk_main () at ../../../../gtk/gtkmain.c:1328
        loop = 0x55555d05ac20
#101 0x00007ffff783a955 in wxGUIEventLoop::DoRun() (this=0x5555557e5120) at ../src/gtk/evtloop.cpp:65
        loopLevel = 0
#102 0x00007ffff735097d in wxEventLoopBase::Run() (this=0x5555557e5120) at ../src/common/evtloopcmn.cpp:78
        activate = {m_evtLoopOld = 0x0}
#103 0x00007ffff73199a6 in wxAppConsoleBase::MainLoop() (this=0x5555557c4e50) at ../src/common/appbase.cpp:334
        mainLoop = {<wxEventLoopBasePtr> = {m_ptr = 0x5555557e5120}, m_pp = 0x5555557c4fe8, m_pOld = 0x0}
#104 0x00005555555ec719 in APP_KICAD::OnRun() (this=<optimized out>) at ./kicad/kicad.cpp:271
        __FUNCTION__ = "OnRun"
#105 0x00007ffff73a1a19 in wxEntry(int&, wchar_t**) (argc=<optimized out>, argv=<optimized out>) at ../include/wx/app.h:439
        initializer = {m_ok = true}
#106 0x00005555555d7198 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at ./kicad/kicad.cpp:302
(gdb) 

```

baryluk, 2020-10-27:
Video with reproduction:

[image]

dbrgn, 2020-10-27:
You're in the wrong issue tracker, this is LibrePCB, not KiCAD :slightly_smiling_face:

closed 2020-10-27 by dbrgn

baryluk, 2020-10-27:
@dbrgn Ooops. I don't know how I got here. Sorry!

kliment, 2020-10-27:
For reference, the kicad issue tracker is at [gitlab](https://gitlab.com/kicad/code/kicad/-/issues)

baryluk, 2020-10-27:
@kliment I found this issue is already reported few days ago.  https://gitlab.com/kicad/code/kicad/-/issues/6102

I just had LibrePCB and KiCAD websites opened at the same time, I didn't pay attention. Sorry again.