LibrePCB/LibrePCB #813: Option for Via tenting

Tubbles, 2020-12-27:
##### SUMMARY

It should be possible to select whether a via should be tented (covered with solder mask) or not. It can be useful for using vias as test points during board debugging.

##### SOLUTION

dbrgn, 2020-12-27:
Is there a standard to indicate tenting/non-tenting in Gerber files? I assume we'd simply add a via-sized hole in the soldermask? (Right now vias simply don't affect the soldermask.)

What about plugged vias?

Tubbles, 2020-12-27:
> Is there a standard to indicate tenting/non-tenting in Gerber files?
No idea, lemme check the specs

> I assume we'd simply add a via-sized hole in the soldermask? (Right now vias simply don't affect the soldermask.)
I think this is how it is usually implemented 

> What about plugged vias?
Good question. I do not think it would make a difference if we'd go with the solution proposed above :)

vicencb, 2022-03-05:
It looks like LibrePCB 0.1.6 tents all vias drilled at 0.5mm or less and doesn't tent bigger ones.
Is that automatic behavior the solution to this issue or manual control is planned?
I would prefer having manual control, but the current behavior is more or less acceptable.

ubruhin, 2022-03-06:
> It looks like LibrePCB 0.1.6 tents all vias drilled at 0.5mm or less and doesn't tent bigger ones.

The value of 0.5mm is configurable under *Board -> Design Rules*:

[image]

>  Is that automatic behavior the solution to this issue or manual control is planned?

I think this global value is good for most use-cases, but sometimes it is desired to manually change tenting for individual vias. So I consider this issue as a proposal to make tenting configurable for individual vias.

ubruhin, 2023-03-28:
Are there any opinions about whether non-tented vias should have a stop mask opening for the whole via, or only for its drill? I.e. 

1. Either stop mask opening = outer via diameter + expansion value
2. Or stop mask opening = drill diameter + expansion value

Generally IMHO variant 2. would make more sense because it is more compact (reducing the area of exposed copper and thus reducing the risk of short circuits between the via and close pads), but so far I guess variant 1. is much more commonly used. One advantage of variant 1. is of course that the via is accessible by a probe, i.e. implicitly acts as a test point.

I wonder if in the end we need 3 different configuration options in LibrePCB:

1. Tent via
2. Drill-only opening
3. Full via opening

But as always, I don't like adding complexity which might actually not be needed in real world scenarios... So any input would be welcome what PCB designers really need :wink:

Maybe it would be good enough to just add a single boolean property to each via to enforce a full opening? I.e.

- [ ] **Full stop mask opening** (unchecked) -> Either tented or drill-only opening, determined by global design rules
- [x] **Full stop mask opening** (checked) -> Enforce full via opening, independent of the global design rules

This would not allow to tent a via which is non-tented by design rules, but maybe that's no use-case anyway. But it might be a bit annoying that by default all vias are not accessible as test points :thinking:

closed 2023-05-27 by ubruhin