LibrePCB/LibrePCB #815: Resizing panes in some editors

Tubbles, 2020-12-27:
##### SUMMARY

In some editors, the compartments where the view panes are placed are not resizable, making them take up a lot of dead space on high resolution monitors.

Specifically I would expect the following pane delimiters (marked red) be resizable:

Device Editor
[image]

Symbol Editor
[image]

Package Editor
[image]

Component Editor
[image]

##### SOLUTION

EdizonTN, 2020-12-27:
I agree!

I wrote by the another words here: https://github.com/LibrePCB/LibrePCB/issues/726

dbrgn, 2020-12-27:
I'd say #726 is a superset of this issue (but they both sound like they might be useful).

This issue could probably be achieved just by fiddling with the layout files.

Tubbles, 2020-12-27:
> I wrote by the another words here: #726

Argh! Was certain I had searched through the backlog, good stuff! :)

> I'd say #726 is a superset of this issue (but they both sound like they might be useful).
> 
> This issue could probably be achieved just by fiddling with the layout files.

It sounds like this issue is superseded by #726, I propose we close this

dbrgn, 2020-12-27:
Well, if you just want resizing then keep it open. #726 would be quite a big refactoring that would require touching most UI code and integrating a new library, while this issue can probably be done just by tweaking UI files. So I still see value in a separate feature request with reduced scope.

Tubbles, 2020-12-27:
Alright, lets keep it then :) I'm not so well versed in Qt but might give it a shot

BuFran, 2023-10-02:
Yeah I searched for that, but I didn't found this issue. I'm sorry for #1244.  

I can make it because it is only simple touching of *.ui files. 

So i'm just asking @ubruhin , it's nice and easy work for my next PR ? :-)

ubruhin, 2025-08-04:
The new UI (see #1494) introduces a very new layout of the editors, and increases the working area in several ways. Therefore maybe this issue is not relevant anymore. Let's first gain some experience with the new UI of LibrePCB 2.0, and if there is still a need to resize panels or so, let's address it in a new issue.

closed 2025-08-04 by ubruhin