LibrePCB/LibrePCB #819: Add support for differential pair routing

Tubbles, 2020-12-28:
##### SUMMARY

Differential pair signalling is a technique to lower the common mode noise of a signal. We should add some quality of life features to make routing these signals easier, such as automatic length matching and simultaneous interactive routing

##### SOLUTION

[Altium's solution](https://www.altium.com/documentation/altium-designer/differential-pair-routing-ad)

Might already be covered with the [kicad router](https://librepcb.discourse.group/t/add-interactive-push-and-shove-router-from-cern-kicad-horizon/)

dbrgn, 2020-12-28:
As a poor-man's-solution, you can measure the length of selected segments by right-clicking and choosing "Measure Selected Segments Length". It's not ideal, but at least it does allow you to match trace lengths.

A dedicated tool for length matching and -tuning would definitely be great!

Tubbles, 2020-12-28:
> ... you can measure the length of selected segments ...

Yeah

Tubbles, 2020-12-28:
Accidental close T_T

z3ugma, 2025-05-05:
I've recently encountered several needs for this! Happy to revive this issue, to add differential pair routing