LibrePCB/LibrePCB #822: Size selectors don't fit their values

Tubbles, 2020-12-28:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.2.0-unstable
Git Revision: 7641b5633f
Qt Version: 5.15.2
Operating System: Manjaro
...
```

##### SUMMARY

The trace and via (and possibly others?) size selectors seems to have changed somewhat, and are no longer capable of displaying their selected values (out of room)

[image]

[image]

##### STEPS TO REPRODUCE

Fire up Board Editor on the master branch

##### EXPECTED RESULTS

Possibility to view the entered value in its entirety

##### ACTUAL RESULTS

The entered value is impossible to view in its entirety

##### ADDITIONAL INFORMATION

dbrgn, 2020-12-29:
Huh, thanks for reporting that _before_ we create the next release :slightly_smiling_face:

ubruhin, 2020-12-30:
Strange, I can't reproduce this even on a similar system (Qt5.15.2 on Manjaro):

[image]

Could you check if the same issue exists with the [nightly AppImage](https://download.librepcb.org/nightly_builds/master/librepcb-nightly-linux-x86_64.AppImage)?

ubruhin, 2024-02-19:
Closing this because of inactivity for quite some time and because I could never reproduce it. If it's still relevant, please provide more details (e.g. used system theme).

closed 2024-02-19 by ubruhin