LibrePCB/LibrePCB #833: Overlapping planes on the same net not considered a connection.

brim2full, 2021-01-22:
##### VERSION / OS / ENVIRONMENT
LibrePCB Version: 0.1.5 and MAIN on GitHub as of 22 Jan 2021
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T20:06:07
Linking type:     static
Unbundled libs:   -
Qt Version:       5.15.0 (built against 5.15.0)
CPU Architecture: x86_64
Operating System: Windows 10 Version 2009
Platform Plugin:  windows

##### SUMMARY
When a trace and plane on the same net are overlapped they are considered as forming a connection.  However overlapping to planes on the same net do not form a connection.

##### STEPS TO REPRODUCE
1. reate a simple circuit consisting of two, two pin connectors and a GND symbol.
2.     Join pins 1 and 2 of one connector to pins 1 and 2 of the second.  Connect pins 2 net to GND.
3.     Place the two connectors on the board. Routing only pins 1 net on the top copper.
4.     Create two separate and non-overlapping planes over each connector on the top copper. And rebuild the planes.  There should still show a air-wire connecting pins 2 net.
5.     Add a trace on the top copper over both planes, do not route to the connectors, a connection is formed.
6.     Remove the trace and partially overlap the planes.  No connection is formed.  <<BUG

##### EXPECTED RESULTS
Overlapping two planes on the same net should function the same as overlapping a plane and a net.

brim2full, 2021-01-25:
A addendum to this issue: 

Two planes should be considered connected if they are on the same net and shares an edge which is within the minimum copper to copper distance or overlap.

Tubbles, 2022-01-27:
Possible duplicate of #562

ubruhin, 2023-01-07:
> Possible duplicate of https://github.com/LibrePCB/LibrePCB/issues/562

Probably yes, thus closing this one in favor of #562.

closed 2023-01-07 by ubruhin