LibrePCB/LibrePCB #836: Library Editor : Cannot place bottom pad at same position of top pad

doom-fr, 2021-01-27:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY
It seems to be impossible to place bottom pad at the same position of top pad in Library Editor

##### STEPS TO REPRODUCE

1. create a new footprint
2. place a pad on the top
3. click on the blue icon to select bottom
4. place a bottom pad at same position
   => can't place it

##### EXPECTED RESULTS
The pad is placed on bottom

##### ACTUAL RESULTS
The pad can't be placed at this position, but it is possible at other positions.

##### ADDITIONAL INFORMATION
workaround, place the pad at another position and move it to the good position.

dbrgn, 2021-01-27:
Confirmed!

doom-fr, 2021-02-06:
In fact the problem is not a problem of layer.
It is also impossible to put a top pad on top of another just placed top pad.

But if you deselect the pad tool and restart it, it works (whatever the layer is).

PR #876 (2021-03-06): Package editor: Stop rejecting but warn about overlapping pads
In the package editor, it was not possible to place two consecutive pads at the same position, see #836. This was strange behavior since after leaving and re-entering the "place pad" tool it was possible. The same applied to the "add hole" tool. To fix this strange behavior, I removed this limitation, so it is now possible to place to consecutive pads or holes at the same position.

However, now it's easier to accidentally place multiple pads at the same position without realizing it. Thus I implemented a package check which shows  [... cut]
closed 2021-03-06 by ubruhin