LibrePCB/LibrePCB #837: Board Editor : Can't grab device and top grab area is not shown

doom-fr, 2021-01-27:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY
There's multiple **very annoying** problems in Board editor.
It seems there's a problem with grab area.

##### STEPS TO REPRODUCE

1. go to board editor
2. activate Top Grab Areas layer
3. click on a device to place => the top grab area is correctly shown in the Place Device window
4. place the device
5. the top grab area is not displayed
6. try to move the device => it can't be moved
7. try to select (rectangle select) the device => it can't be selected (the rectangle of selection is not drawn during selection)
8. close board editor
9. open it back => still can't move and can't select
10. close board editor & close project
11. open the project, go to board editor
12. try to select => it works
13. try to move => it works

##### EXPECTED RESULTS

- top grab area should be displayed
- we should be able to select
- we should be able to move

##### ACTUAL RESULTS
See Step to reproduce

##### ADDITIONAL INFORMATION
It is on a device from a local library.
It has been drawn in multiple steps (forget to draw top grab area at the beginning).

doom-fr, 2021-01-27:
[prog_v2.lppz.zip](https://github.com/LibrePCB/LibrePCB/files/5877365/prog_v2.lppz.zip)

doom-fr, 2021-01-27:
[image]

ubruhin, 2021-01-27:
I think this is a duplicate of #787 which was fixed in #794. Can you confirm it works properly when manually activating the "select" tool after placing the device?

doom-fr, 2021-01-27:
Yes, I confirm, manually activating the "select" tool make it working (for select and move).
So yes for this part, it is a duplicate of issue #787.

But why the top grab area is not displayed ?

I have drawn a rectangle in "top grab areas" layer with checkboxes "filled" & "grab area" checked.

ubruhin, 2021-01-28:
> So yes for this part, it is a duplicate of issue #787.

OK, perfect :+1:

> I have drawn a rectangle in "top grab areas" layer with checkboxes "filled" & "grab area" checked.

You mean the "Top Hidden Grab Areas" layer? If yes, this layer is *not* visible in the board editor (that's why it is called *hidden* :wink:). Actually, the idea is to just enable the "grab area" checkbox on some outline polygon (e.g. on layer "Top Placement"). However, since with the grab areas it's much harder to select e.g. vias or traces below/behind a device, usually we don't use the "grab area" feature in footprints. You can grab any device on its origin (coordinate 0,0) anyway...

ubruhin, 2021-03-06:
Closing this as it is a duplicate of #787.

closed 2021-03-06 by ubruhin