LibrePCB/LibrePCB #838: Gerber : Mysterious message "Value must be > 0 !"

doom-fr, 2021-01-27:
##### SUMMARY

[image]

"Value must be > 0 !" => Which value ?
(think it was hole size)

##### SOLUTION
Display a more specific message : tell the value that need to be >0

ubruhin, 2021-01-27:
The error comes from a very low level class which doesn't "know" what it is used for (could be a hole diameter, trace width, line length, ...) thus it's hard to say where the actual error is located... But yeah, most probably there is some hole with a diameter <=0. Can you confirm that? And do you have an idea how it was possible to create such an invalid object (it should actually not be possible)?

doom-fr, 2021-01-27:
Yes it is related to #839 !

ubruhin, 2023-02-21:
I think with the refactoring done in #1076 this invalid case can no longer exist. Holes with a diameter of 0mm will be removed during the file format migration 0.1->0.2 and cannot be created anymore with 0.2.x. So let's close this issue.

closed 2023-02-21 by ubruhin