LibrePCB/LibrePCB #839: Library Editor : THT without hole

doom-fr, 2021-01-27:
##### SUMMARY
[image]

What's the way to place top and bottom pad without hole ?

When we put hole size = 0mm, DRC warns and gerber generation report an error.

Placing a top and a bottom pad is possible but it would be better to have another option.

##### SOLUTION

Manage 0mm hole size as "no hole" or add a "no hole" option.

dbrgn, 2021-01-27:
A through-hole pad without a hole is arguably not a through-hole pad.

Why don't you add two SMD pads, one on top and one at the bottom?

doom-fr, 2021-01-27:
Obviously I tried, but it is more difficult.
And I encounter two bugs (#836 & #840), so really not easy.

If it is not allowed to do THT with zero size hole, we need to prevent to do it.

dbrgn, 2021-01-28:
Yes, that's true, the drill diameter should not allow 0 to be entered!

ubruhin, 2023-01-15:
This is actually fixed with #1076 which refactored the concept of pad holes :muscle:

closed 2023-01-15 by ubruhin