LibrePCB/LibrePCB #840: Board Editor : Systematic crash when saving after a "no element of type pad with UUID"

doom-fr, 2021-01-27:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY

I had a problem with zero size hole in a footprint I created.
So I decided to change from THT zero size hole to one top and one botom pad.
I changed 'A' pad to 'A1' and set it as top pad, create also a pad 'A2' and set it as bottom pad.
I done the same with 'B' pad.
See also that there were unused pad 5 & 6, so remove them.
When I tried to update the footprint, I experienced several issues.
1 - it does not update in the board editor (do not remember if it is normal or not).
2 - when I suppress and replace the device, it shows an error
[image]
3 - when I save it crashes. The crash is systematic.

##### STEPS TO REPRODUCE

1. open LibrePcb
2. open Board Editor
3. try to place the updated device => UUID error
4. try to save => systematic crash

##### EXPECTED RESULTS
It should work :-)

##### ACTUAL RESULTS
It crashes.

##### ADDITIONAL INFORMATION

Console output :

```
./librepcb-0.1.5-linux-x86_64.AppImage 
Load stroke font: "newstroke.bene"
Start loading font "/tmp/.mount_librepmW6PRE/opt/share/librepcb/fontobene/newstroke.bene"
[  INFO   ] "LibrePCB 0.1.5 (a435c5095c6f94df4da130fe832ea97f42d17901)" (:0)
[  INFO   ] "Qt version: 5.14.2 (compiled against 5.14.2)" (:0)
[  INFO   ] "Resources directory: /tmp/.mount_librepmW6PRE/opt/share/librepcb" (:0)
[  INFO   ] "Application settings: /home/doom/.config/LibrePCB/LibrePCB.ini" (:0)
[DEBUG-MSG] Started network access manager thread. (:0)
[ WARNING ] There was a stale lock on the workspace: "FilePath(/doompath/LibrePCB-Workspace)" (:0)
[DEBUG-MSG] Load workspace settings... (:0)
[DEBUG-MSG] Workspace settings loaded. (:0)
[DEBUG-MSG] Load workspace library database... (:0)
[DEBUG-MSG] Workspace library database successfully loaded! (:0)
[DEBUG-MSG] Workspace library scanner thread started. (:0)
[DEBUG-MSG] Workspace library scan started. (:0)
[DEBUG-MSG] Successfully loaded font "/tmp/.mount_librepmW6PRE/opt/share/librepcb/fontobene/newstroke.bene" with 2573 glyphs (:0)
[DEBUG-MSG] Workspace libraries indexed: 9 libraries in 17 ms (:0)
[DEBUG-MSG] open project: "/doompath/LibrePCB-Workspace/projects/prog_v2/prog_v2.lpp" (:0)
[DEBUG-MSG] Load stroke font: "newstroke.bene" (:0)
[DEBUG-MSG] Start loading font "/doompath/LibrePCB-Workspace/projects/prog_v2/resources/fontobene/newstroke.bene" (:0)
[DEBUG-MSG] load project metadata... (:0)
[DEBUG-MSG] metadata successfully loaded! (:0)
[DEBUG-MSG] load settings... (:0)
[DEBUG-MSG] settings successfully loaded! (:0)
[DEBUG-MSG] load project library... (:0)
[DEBUG-MSG] successfully loaded 4 symbols (:0)
[DEBUG-MSG] successfully loaded 3 packages (:0)
[DEBUG-MSG] successfully loaded 4 components (:0)
[DEBUG-MSG] successfully loaded 3 devices (:0)
[DEBUG-MSG] project library successfully loaded! (:0)
[DEBUG-MSG] load circuit... (:0)
[DEBUG-MSG] circuit successfully loaded! (:0)
[DEBUG-MSG] 1 schematics successfully loaded! (:0)
[DEBUG-MSG] Successfully loaded font "/doompath/LibrePCB-Workspace/projects/prog_v2/resources/fontobene/newstroke.bene" with 2573 glyphs (:0)
[DEBUG-MSG] 1 boards successfully loaded! (:0)
[DEBUG-MSG] project successfully loaded! (:0)
[DEBUG-MSG] Workspace library scan succeeded: 4928 elements in 10656 ms (:0)
[EXCEPTION] There is no element of type "pad" with the UUID "1d4cc8ea-856c-437a-8fbf-99371158f18d" in the list. (../../../../libs/librepcb/common/geometry/../fileio/cmd/../serializableobjectlist.h:449)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
Erreur de segmentation (core dumped)
```

doom-fr, 2021-01-27:
I tried to remove pads and creating them back.
But this doesn't save me.
[image]

doom-fr, 2021-01-27:
Found this in the component ...

[image]

doom-fr, 2021-01-27:
So that's the solution.

1. Modify the footprint
2. Open the component
3. Change to another footprint
4. Change back to the good footprint
5. Assign component signal to package pad

This is the what we get after : 
[image]

doom-fr, 2021-01-27:
I think this is related to issue https://github.com/LibrePCB/LibrePCB/issues/552

doom-fr, 2021-01-27:
[image]

doom-fr, 2021-01-28:
This is a bug (obviously segfault is a bug)
So @ubruhin @dbrgn, can you please add "Bug" tag to this issue ?

ubruhin, 2021-01-29:
Confirmed! Thanks for the detailed bug report :+1:

doom-fr, 2021-01-29:
Thanks. And sorry to add work for the next release :smile:

PR #855 (2021-02-14): Fix crash after adding a device to a board failed
When adding a new device to a board failed (for example due to invalid library elements), some objects were not cleaned up properly which caused a segfault some time later (for example when saving the project). Now these objects are cleaned up to fix this issue.

Fixes #840
closed 2021-02-14 by ubruhin