LibrePCB/LibrePCB #841: Schematic Editor : error while pasting : "net signal cannot be changed because it is still in use"

doom-fr, 2021-01-31:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY

[image]

##### STEPS TO REPRODUCE

1. cut
2. paste

##### EXPECTED RESULTS

It should paste correctly

##### ACTUAL RESULTS

It shows an error and is not pasted.
(and the source does not come back)

##### ADDITIONAL INFORMATION

hopefully, crtl+Z works

doom-fr, 2021-01-31:
I tried to delete the wire on Ub:GND, but I got another error :  :sob: 

[image]

ubruhin, 2021-02-01:
I cannot reproduce this. Could you show a screenshot of the schematic where we can see which part of it you cut/pasted?

doom-fr, 2021-02-01:
[image]

doom-fr, 2021-02-01:
Here is the part of the schematic I want to cut and paste.
I don't really understand the message error.
What does it mean ? What does mean "it is still in use" ?

doom-fr, 2021-02-01:
Here is the message if I try to make a copy/paste instead of a cut/paste :
 
[image]

doom-fr, 2021-02-05:
I encountered the same message when connecting pins of a component 
[image]

Followed by a second error : 
[image]

It was on a symbol with multiple pins assigned to the same component signal.

It was not when copy/cut/paste but when simply drawing a wire.

doom-fr, 2021-02-05:
The error occurs only when starting the wire on another component and ending it on the "multiple" pad.
If starting from the "multiple" pad, there's no error.

doom-fr, 2021-02-05:
Also occurs when trying to suppress wire.
Always related to "multiple" type pin.