LibrePCB/LibrePCB #843: Library Editor : Allow to create symbol using an image to show the internal circuitry

doom-fr, 2021-02-01:
##### SUMMARY

For some specialized component with integrated features, it is great to see what's inside when drawing the schematic.
Unfortunately, it is time consuming to draw each element in the symbol.
It is usually already drawn in the datasheet, so it will be great to use this already done job.

##### SOLUTION
Allow to paste an image in library editor with the internal circuitry of the component.
To help keeping the schematic pretty, maybe we could only allow black and white image (it is usually the case in a datasheet) and make a little image processing to change : 
-  the white -> transparent
- the black -> the same colour used in the schematic

There might be an issue with licence of the drawing ...

dbrgn, 2021-02-01:
Yes, licensing, plus the image will look crappy when scaled, because datasheets usually contain raster graphics while the schematic editor contains vector graphics.

I don't think this is a good idea in the long term.

doom-fr, 2021-02-06:
With PDF format, it is usually vector graphics.
But licence will be a problem.
There's many other features to develop, so I close this issue.

closed 2021-02-06 by doom-fr