LibrePCB/LibrePCB #848: LibrePCB : Auto add line on 'Enter'

doom-fr, 2021-02-06:
##### SUMMARY

LibrePCB uses a lot fields in table with a '+' at the end of the line to validate the line.

[image]

It is important to click on this '+'. In other cases, what you just entered is not saved.

So first, it a source of mistake, to miss to click on '+'

And also, we loose time to go from keyboard to mouse to click the '+'.
(obviously when you do it a lot of time)

##### SOLUTION

When we are filling a new line and we press 'enter' to validate the current field,
if we press 'enter' a new time, it should add the line

If next one is 'tab', we start to fill a new line, if it was 'enter', then it close the window.

EdizonTN, 2021-02-06:
I highly recommended, **entering these fields without using a mouse**! 
Especially in Pin/Signals definitions (ex. in more pins IC!).

Please, try to apply logic like in the Excel:
- Tab to enter the field
- simple Enter as finish editing the field and move to next field (activate for editing - don't clear content if exists!)
- CTRL+Enter for multiline fields as new line in the field (if this type exists)
- cursor keys as navigate inside active field or through fields. 
 

If I press enter in the last field in row, create a new one (same as is '+' pressed). If created row's fields was not edited, whole row will be ignored.
Don't close windows using enter key. This is little bit confusing. 
I think, nothing better doesn't exist.

dbrgn, 2021-02-07:
I agree, and if an unsaved row is present, closing the window should prompt the user, instead of just losing the data.

closed 2023-08-06 by ubruhin

doom-fr, 2023-08-28:
I agree what @dbrgn said about unsaved rows.
You also @ubruhin :wink: 

Unfortunately, it seems it has not been implemented in 1.0.0-r1 :sob: 

And If I can choose, I prefer the line to be validated without any prompt (except if it fails).

ubruhin, 2023-08-28:
Hm, but unsaved rows should no longer be an issue since Enter does *not* close the window anymore while editing a table row? So IMHO it is now very unlikely that one closes a dialog by accident.

> And If I can choose, I prefer the line to be validated without any prompt (except if it fails).

If you mean the line should be added automatically: I didn't implement that because with the current representation this behavior would not be intuitive. In many tables there is more than one column, but the columns 2..n would not make sense on the last row if the data is committed immediately after modifying column 1.

Anyway, I plan to completely replace this UI by something new soon :slightly_smiling_face:

doom-fr, 2023-08-28:
I think it's more intuitive to auto-add than to dismiss without saying anything.
(for the user, it's counterintuitive that what he see in the table will be dismissed without notification)

The "OK" button can still be clicked or activated by CTRL+Enter.
Maybe at least, prompt something to say there's a not-validated line.

But obviously, if you replace it by something new soon, maybe this can wait.

PR #1184 (2023-08-06): Support enter/delete shortcuts in table widgets
In most table widgets, the keyboard shortcut *Enter* now adds a new row (as requested in #848) and *Delete* now removes the selected row. However, *Enter* often needs to be pressed twice because while a cell is in edit mode, the first *Enter* exits the edit mode (so you could move the focus to another table cell then).

Usability is much better now especially because *Enter* first adds a new row instead of (accidentally) closing a modal dialog.

Closes #848.