LibrePCB/LibrePCB #853: Board Editor : Crash when updating library (SegFault)

doom-fr, 2021-02-13:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY
it crash during library update

##### STEPS TO REPRODUCE

1. open board editor
2. try to place a just created device
3. get an error
[image]
"Could not add device:
There is no element of type "pad" with the UUID "230551dd-ac8c-40e4-8032-907d31c9954e" in the list"
4. update library

##### EXPECTED RESULTS

all goes well

##### ACTUAL RESULTS

it crashes : 

```
doom@doom:~/software$ ./librepcb-0.1.5-linux-x86_64.AppImage 
Load stroke font: "newstroke.bene"
Start loading font "/tmp/.mount_librepNgmZT5/opt/share/librepcb/fontobene/newstroke.bene"
[  INFO   ] "LibrePCB 0.1.5 (a435c5095c6f94df4da130fe832ea97f42d17901)" (:0)
[  INFO   ] "Qt version: 5.14.2 (compiled against 5.14.2)" (:0)
[  INFO   ] "Resources directory: /tmp/.mount_librepNgmZT5/opt/share/librepcb" (:0)
[  INFO   ] "Application settings: /home/doom/.config/LibrePCB/LibrePCB.ini" (:0)
[DEBUG-MSG] Started network access manager thread. (:0)
[DEBUG-MSG] Load workspace settings... (:0)
[DEBUG-MSG] Workspace settings loaded. (:0)
[DEBUG-MSG] Load workspace library database... (:0)
[DEBUG-MSG] Workspace library database successfully loaded! (:0)
[DEBUG-MSG] Workspace library scanner thread started. (:0)
[DEBUG-MSG] Workspace library scan started. (:0)
[DEBUG-MSG] Successfully loaded font "/tmp/.mount_librepNgmZT5/opt/share/librepcb/fontobene/newstroke.bene" with 2573 glyphs (:0)
[DEBUG-MSG] Workspace libraries indexed: 9 libraries in 11 ms (:0)
[DEBUG-MSG] Restoring file system from autosave backup: "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/.autosave/autosave.lp" (:0)
[DEBUG-MSG] open project: "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/Rbv3.lpp" (:0)
[DEBUG-MSG] Load stroke font: "newstroke.bene" (:0)
[DEBUG-MSG] Start loading font "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/resources/fontobene/newstroke.bene" (:0)
[DEBUG-MSG] load project metadata... (:0)
[DEBUG-MSG] metadata successfully loaded! (:0)
[DEBUG-MSG] load settings... (:0)
[DEBUG-MSG] settings successfully loaded! (:0)
[DEBUG-MSG] load project library... (:0)
[DEBUG-MSG] successfully loaded 63 symbols (:0)
[DEBUG-MSG] successfully loaded 0 packages (:0)
[DEBUG-MSG] successfully loaded 54 components (:0)
[DEBUG-MSG] successfully loaded 0 devices (:0)
[DEBUG-MSG] project library successfully loaded! (:0)
[DEBUG-MSG] load circuit... (:0)
[DEBUG-MSG] circuit successfully loaded! (:0)
[DEBUG-MSG] 4 schematics successfully loaded! (:0)
[DEBUG-MSG] Successfully loaded font "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/resources/fontobene/newstroke.bene" with 2573 glyphs (:0)
[DEBUG-MSG] 1 boards successfully loaded! (:0)
[DEBUG-MSG] project successfully loaded! (:0)
[DEBUG-MSG] closed project: "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/Rbv3.lpp" (:0)
[EXCEPTION] Autosave backup detected in directory '/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3'. (../../../libs/librepcb/common/fileio/transactionalfilesystem.h:122)
[DEBUG-MSG] Workspace library scan succeeded: 5046 elements in 11108 ms (:0)
[DEBUG-MSG] Restoring file system from autosave backup: "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/.autosave/autosave.lp" (:0)
[DEBUG-MSG] open project: "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/Rbv3.lpp" (:0)
[DEBUG-MSG] Load stroke font: "newstroke.bene" (:0)
[DEBUG-MSG] Start loading font "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/resources/fontobene/newstroke.bene" (:0)
[DEBUG-MSG] load project metadata... (:0)
[DEBUG-MSG] metadata successfully loaded! (:0)
[DEBUG-MSG] load settings... (:0)
[DEBUG-MSG] settings successfully loaded! (:0)
[DEBUG-MSG] load project library... (:0)
[DEBUG-MSG] successfully loaded 63 symbols (:0)
[DEBUG-MSG] successfully loaded 0 packages (:0)
[DEBUG-MSG] successfully loaded 54 components (:0)
[DEBUG-MSG] successfully loaded 0 devices (:0)
[DEBUG-MSG] project library successfully loaded! (:0)
[DEBUG-MSG] load circuit... (:0)
[DEBUG-MSG] circuit successfully loaded! (:0)
[DEBUG-MSG] 4 schematics successfully loaded! (:0)
[DEBUG-MSG] Successfully loaded font "/home/doom/electro-HDR/LibrePCB-Workspace/projects/Rbv3/resources/fontobene/newstroke.bene" with 2573 glyphs (:0)
[DEBUG-MSG] 1 boards successfully loaded! (:0)
[DEBUG-MSG] project successfully loaded! (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] There is no element of type "pad" with the UUID "230551dd-ac8c-40e4-8032-907d31c9954e" in the list. (../../../../libs/librepcb/common/geometry/../fileio/cmd/../serializableobjectlist.h:449)
Erreur de segmentation (core dumped)
```

##### ADDITIONAL INFORMATION

doom-fr, 2021-02-13:
It was because of this :

```
  (pad 230551dd-ac8c-40e4-8032-907d31c9954e (side tht) (shape round)
   (position 0.0 7.35) (rotation 0.0) (size 2.0 2.0) (drill 1.3)
  )
```

I think it comes from a suppression of pad in library.
It might not be well suppressed.

ubruhin, 2021-02-13:
Hi @doom-fr, isn't this a duplicate of #840? :wink:

doom-fr, 2021-02-14:
Not sure but maybe.
It is not exactly the same situation, but it seems to come from the same part of code (serializableobjectlist.h:449)

doom-fr, 2021-02-14:
And also, I observe this after trying to solve the problem : 
[image]

doom-fr, 2021-02-14:
I changed to other device and save.
This solves this UUID display in pre-selected field. :crossed_fingers:

ubruhin, 2021-02-14:
OK I see, it seems the only difference is that here it crashes when updating the library, and in #840 it crashes when saving the project. This makes totally sense since during library update, the project is saved as well. So probably this is indeed exactly the same issue. I implemented the bugfix in #855 so this should be fixed in the next release.

The thing with displaying an UUID instead of a device name is not related to this issue, it just means that the selected device is not found anymore in the workspace library, which is expected behavior and does not cause any issues.

Let's close this issue as it seems to be a duplicate of #840. Feel free to reopen if it still exists in the next release.

closed 2021-02-14 by ubruhin