LibrePCB/LibrePCB #861: Add distance measurement tool

ubruhin, 2021-02-21:
##### SUMMARY

A simple tool to measure distances in all editors.

##### SOLUTION

Maybe as a new item in the "Tools" toolbar. The first mouse click sets the start point of the measurement, the second click sets the end point and somehow the distance should be displayed (without a message box would be nice).

Would be nice to snap to underlying graphics elements (e.g. center of pads) by default, and disable snapping by pressing the SHIFT modifier (like the trace drawing tool).

dbrgn, 2021-02-21:
Would be good if the tool wouldn't just measure direct (cartesian) distance, but also horizontal and vertical distance for the drawn line.

doom-fr, 2021-02-21:
It will be great to allow measurement of the width of a drawn segment.
(but it will be better to get this info with a right click on the segment)

dbrgn, 2021-02-21:
> (but it will be better to get this info with a right click on the segment)

That's already possible. Right-click on a trace segment, then select "Measure".

doom-fr, 2021-02-21:
> That's already possible. Right-click on a trace segment, then select "Measure".

This is the length, not the width.

dbrgn, 2021-02-21:
Ah, I mis-read your message! I guess showing the width of a trace would be part of a feature, where the width of a traace can be changed (#735).

doom-fr, 2021-02-21:
Ah, we really need 0.2.0 ! :wink:

closed 2022-09-03 by ubruhin