LibrePCB/LibrePCB #863: Board Editor : Impossible to connect traces on pads

doom-fr, 2021-02-24:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY

The connection of pads usually works well but it becomes impossible to connect pads.

##### STEPS TO REPRODUCE

Do not really know where it comes from.

##### EXPECTED RESULTS

Trace is correctly connected to pad

##### ACTUAL RESULTS

[image]

##### ADDITIONAL INFORMATION

doom-fr, 2021-02-24:
It seems it comes from the fact I used the 'shift' key at the precedent trace drawing.
Maybe something is not cleaned correctly.
Or my 'shift' key on my keyboard was blocked ? But it is quite new and it's a good one (Dell), so I don't think so.

Not able to reproduce.

doom-fr, 2021-02-24:
> Or my 'shift' key on my keyboard was blocked ?
I don't think so as I wrote the issue between and all is not written in CAPITALS.

ubruhin, 2023-02-21:
"Unfortunately" I was never able to see this problem. Probably it doesn't exist anymore in the mean time since in 0.1.6 and 0.1.7 there were a lot of improvements made in the corresponding logic. Thus I'll close this issue, but of course please reopen if the problem still occurs.

closed 2023-02-21 by ubruhin