LibrePCB/LibrePCB #864: Board Editor : For THT pad, add copper ring only for top and bottom and used inner layer

doom-fr, 2021-02-25:
##### SUMMARY

THT pads adds copper ring for top, bottom, but also for all inner layers and sometimes it is not useful.

For example, for this kind of circular right angle pcb mount connector, the only way to connect the center pin is to use inner layer, but because of the generated ring on the inner layers, it produces a DRC, and there's a real spacing problem.
[image]

##### SOLUTION

- Only generate ring for inner layers if the pad is connected via this inner layer
or
- Allow to select on which layer the ring is generated.

Maybe it can also be generalized to via.

[image]

ubruhin, 2023-02-06:
I was working on this topic. Unfortunately it's not easy to understand under which conditions it's safe to completely omit inner annular rings and what design rules would need to be fulfilled. Since drilling is not perfectly accurate, the drill might have an offset and could create a short circuit with other copper objects (e.g. traces) which are placed close to the pad drill (too close, since the annular ring is missing).

Also planes would need to calculate clearances to pads without annular ring differently and I'm not sure where and how this clearance parameter should be defined (on the plane? globally? ...?)

After some thinking about this, I came up with a quite simple solution without completely omitting annular rings, but still allowing much more compact designs. See #1097 for details.

Basically you can now set the minimum annular ring to a value which is just small enough to avoid creating any short circuits due to drill tolerances. Then you'll get very small annular rings on unconnected layers - these annular rings can be considered as the "keepout zone" for any other copper on that layer. This way, the plane clearances and DRC rules work as usual (no special logic needed for omitted pads).

@doom-fr any opinion about this solution? Would it solve your problem? Note that currently it is a global configuration, of course one day we probably also want to override it only for particular pads - but let's ignore that for the moment (step by step) :wink:

ubruhin, 2023-02-13:
I'll close this for now since #1097 is merged. Feel free to comment or reopen if you're not happy with the implemented solution :)

closed 2023-02-13 by ubruhin