LibrePCB/LibrePCB #869: Board Editor : allow partial DRC checking

doom-fr, 2021-02-27:
##### SUMMARY

DRC checking is a bit long as it checks a lot of things.
(on my current board, more than 40s with a recent laptop)
It will be great to check only what the user needs at the current moment to speed up the process.

##### SOLUTION

On the left of each line of rules, add a checkbox to validate/invalidate this rule checking.
Add also "activate all rules" and "deactivate all rules" buttons.

doom-fr, 2021-07-16:
It will be great also if there's the possibility to check only already found errors.

ubruhin, 2021-08-28:
Implemented in #907, any feedback welcome :slightly_smiling_face: 

> It will be great also if there's the possibility to check only already found errors.

Although that would indeed be nice, currently it would be very difficult to implement - not worth the effort I think. Thus I didn't implement this yet.

closed 2021-09-04 by ubruhin