LibrePCB/LibrePCB #870: Gerber Generation : "Invalid coincident draw" error on Ucamco gerber viewer

doom-fr, 2021-02-28:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY

In some conditions, the generated Gerber files can't be correctly read by the Ucamco Gerber viewer https://gerber-viewer.ucamco.com/
Is seems to be related to cut-out in planes.

##### STEPS TO REPRODUCE

1. generate gerber file and  upload in Ucamco viewer

##### EXPECTED RESULTS

No error

##### ACTUAL RESULTS

[image]

##### ADDITIONAL INFORMATION

###### Error case : 

[image]

###### Good case :

[image]
 
###### Gerber files

[gerber--prob-cutout-planes.zip](https://github.com/LibrePCB/LibrePCB/files/6057978/gerber--prob-cutout-planes.zip)

doom-fr, 2021-02-28:
On another layer (inner2) : 

This is good : 
[image]

And this is bad : 
[image]

[image]

It is on a via, so not J30.

What can be observed is that the via and J30 are very close.

Here are the gerber files : 

[gerber--inner2-planes-prob.zip](https://github.com/LibrePCB/LibrePCB/files/6058181/gerber--inner2-planes-prob.zip)

ubruhin, 2021-03-11:
Thanks for reporting this! I did not find any issue in your Gerber file, so I asked Ucamco about this problem. They say this warning seems to be triggered by a very short line segment in the Gerber file. But there is nothing wrong with the file, it's an issue of their online viewer. They might fix it one day.

So it's safe to hand over such Gerber file to PCB fabricators :slightly_smiling_face: Let's close this issue since we don't need to take care of this.

closed 2021-03-11 by ubruhin