LibrePCB/LibrePCB #873: Gerber Generation : Octogons were not correctly handled by manufacturers

doom-fr, 2021-03-02:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:48:36
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY

2 big manufacturers (PCBWay or JLCPCB) have difficulties to handle Gerber files generated by LibrePCB.
It seems that octagons are not correctly handled.
It is with octagons with difference between width and height.
Others seems to be correctly handled.

##### STEPS TO REPRODUCE

1. create a component with top octagonal pad of 1.0 x 0.6 mm
2. insert it in your design
3. generate Gerber files
4. test it in [Ucamco reference Gerber viewer](https://gerber-viewer.ucamco.com/) => it's ok
5. send them to a manufacturer (here : PCBWay or JLCPCB) => files refused

##### EXPECTED RESULTS

[image]

##### ACTUAL RESULTS

[image]

[image]

##### ADDITIONAL INFORMATION

doom-fr, 2021-03-02:
Check it, but I think this is not a bug of LibrePCB, but a bug in the manufacturer software.

If yes, it will be great to warn the user when drawing octagonal pads, and to advice him to use rounded instead.
It might be also an issue with custom shape pads (#540)

Maybe also octagons pads might be generated not as aperture (option).

doom-fr, 2021-03-03:
I have inspected gerber files and these two lines is the problem : 

```
%ADD33ROTATEDOCTAGON,1.0X0.6X0.175736X90.0*%
%ADD40ROTATEDOCTAGON,1.0X0.6X0.175736X0.0*%
```
I assume that if there's rotated octagon, there should be simple octagons.

A solution might be to generate simple octagons and switch width & height if they are rotated by 90°.

doom-fr, 2021-03-03:
Maybe this can help also : 
http://techservinc.com/pipermail/peda_techservinc.com/2006-November/003210.html

dbrgn, 2021-03-03:
@doom-fr can you share gerber files, or are they secret?

Here's a repro gerber file with 3 non-rotated pads: [OctagonRepro_COPPER-TOP.gbr.txt](https://github.com/LibrePCB/LibrePCB/files/6075502/OctagonRepro_COPPER-TOP.gbr.txt)

And here with 90° rotated pads: [OctagonRepro_COPPER-TOP.gbr.txt](https://github.com/LibrePCB/LibrePCB/files/6075512/OctagonRepro_COPPER-TOP.gbr.txt)

dbrgn, 2021-03-03:
So if I read this correctly, in the non-rotated version, first a custom aperture shape is defined with the `%AM` (aperture macro) command. Then, it is used in the aperture definition (`%AD`) command.

```
%AMROTATEDOCTAGON*4,1,8,-($1/2),($2/2)-$3,-($1/2)+$3,($2/2),($1/2)-$3,($2/2),($1/2),($2/2)-$3,($1/2),-($2/2)+$3,($1/2)-$3,-($2/2),-($1/2)+$3,-($2/2),-($1/2),-($2/2)+$3,-($1/2),($2/2)-$3,$4*%
%ADD11ROTATEDOCTAGON,2.5X1.3X0.380761X0.0*%
```

Is support for aperture macros bad in general? Or maybe only the use of variables is problematic? (I guess we could inline those.)

> I assume that if there's rotated octagon, there should be simple octagons.

No, Gerber does not define octagon shapes. We'll always need custom apertures for those. (Or polygon areas, but that would be terrible.)

doom-fr, 2021-03-03:
Ah ok.
Maybe it will help to see how others CAD software handles this to avoid the problem.

ubruhin, 2021-03-06:
I checked how KiCad 5 and Eagle 7 export such pads. Actually I guess they don't even support octagon pads where width!=height (really?!) so I checked it with rotated oblongs and rotated regular octagons. The result quite surprises me. Kicad uses G36/G37 regions (which is not recommended by the Gerber specs) and Eagle even draws a lot of strokes to fill a pad :joy:

[image]

doom-fr, 2021-03-06:
Different ways to work around the CAM software problem.

Does Kicad uses G36/G37 only for octagons or for all pads ?

And for information, it seems that oblong is not a problem for CAM software.

atommann, 2021-03-11:
Urban and me figured out that the hardcoded octagon Aperture Macro works in a CAM tool (we call it CAM tool A in this post), this makes me belive this bug can be found, another reason I believe it can be fixed is Gerber's Aperture Macro Primitive outline (code 4 in the specification) is a simple one, and it is widely used in Gerber files for various PCB design softwares.

Because the CAM tools used are aged and Gerber spec is always keeping evolving. So one idea is to have a look at the the outline definition in the old Gerber specification as the CAM developers must read the spec when they developed it. I googled and download a copy of version from 2001 (Part Number 414 100 014 Rev D March, 2001, 61 pages in total), there's no big differences between the old one and the pretty new one (Revision 2021.02, 203 pages in total).

In the old spec, on page 20, section "AM parameter syntax rules", it writes:

>Do not begin a variable primitive modifier with a minus sign (for example,
-$1). To indicate negative, precede the variable with 0 (for example, 0-$1).

With the information above, I modified the macro definition, use 0-$1 format, it didn't work.

While in the lates spec, on page 68, it has a section "4.5.4 Macro Variables and Expressions" which is informative and gives lots of examples which is quite good for intuitive learning especially for a person who is not very familiar with Gerber format.

If the hardcoded version works, one natural debugging idea is: Maybe we can try to use the mixed version, i.e. some parameters are hardcoded, but for some of them we use variables. We replace the hardcoded value one by one, see when the macro will trigger the bug of the CAM tool (The only thing we need is just some elbow grease and be patient enough)

The test file is based on the Gerber from @dbrgn.

```
%FSLAX66Y66*%
%MOMM*%
G01*
G74*
%AMROTATEDOCTAGON*4,1,8,-$1/2,$2/2-$3,-0.869239,0.65,0.869239,0.65,$1/2,0.269239,1.25,-0.269239,0.869239,-0.65,-0.869239,-0.65,-1.25,-0.269239,-1.25,0.269239,$4*%
%ADD10ROTATEDOCTAGON,2.5X1.3X0.380761X0.0*%
D10*
X5606051Y5606051D03*
M02*
```

With the above macro, we can get this image in CAM tool A:

[image]

Now if we change ```$2/2-$3``` to ```0.65-$3``` ($2/2 is 0.65), we get:

[image]

This is the shape we expected! How?!

Then I use ```0.5x$2-$3``` and ```$2x0.5-$3```, it didn't helped.

Every corner is OK except the spike at top left, the coordinate of that point must come from the calculation of ```$2/2-$3```, $2 is 1.3, $3 is 0.380761, so I measured the distance between the center of the macro and the spike, the value is around 0.8mm. How can we get 0.8 from ```$2/2-$3```? After several times of attempts I see 1.3/(2-0.380761) = 0.80, is this a coincidence? Maybe.

When I googled the keyword "gerber macro arithmetic operator" I found the following thread and was inspired.

[[Gerbv-devel] [ gerbv-Bugs-3172530 ] Operator precedence not implemented in aperture macros](https://www.mail-archive.com/gerbv-devel@lists.sourceforge.net/msg00318.html)

It's not hard to verify this guess: just change the value of $2 then measure the coordicate of the spike again, if the distance is the same with the value from ```$2/(2-$3)```, then the guess is correct. If ```$2 = 2```, the calculated value is about 1.23. Modify the Gerber accordingly and open it in CAM tool A.

[image]

Yes, it is 1.23!

The bug is clear, let's compose another version for this macro, read the Gerber spec and checked an example macro from KiCad, I get this one (It is a complete Gerber layer, can be saved as a filename.gbr file to test):

```
%FSLAX66Y66*%
%MOMM*%
G01*
G74*

%AMROTATEDOCTAGON*
0 Oblong octagon, 8 corners, with rotation*
0 The origin of the aperture is its center*
0 number of corners: always 8*
0 $1: Length*
0 $2: Width*
0 $3: Chamfer*
0 $4 Rotation angle, in degrees counterclockwise*
0 create outline with 8 corners*
$5=$1/2*
$6=$2/2*
4,1,8,
0-$5,$6-$3,
$3-$5,$6,
$5-$3,$6,
$5,$6-$3,
$5,$3-$6,
$5-$3,0-$6,
$3-$5,0-$6,
0-$5,$3-$6,
0-$5,$6-$3,
$4*
%

%ADD10ROTATEDOCTAGON,2.5X1.3X0.380761X0.0*%
D10*
X0Y0D03*
M02*
```

It works correctly in:

- CAM tool A
- Gerbv
- https://tracespace.io/view/
- Ucamco's reference Gerber viewer

[image]

So I think for compatiblity reason, LibrePCB can consider to modify the this aperture macro definition (note that my version is just a quick and dirty fix). This is not a problem of LibrePCB, it's just the crappy CAM tools caused these issues.

Interestingly, the version works in many Gerber readers does not work in JLCPCB.com's previewer! I've told the relevant developer at JLCPCB to check if it's a bug in their code.

Personally I had fun during this debugging process, I want to thank this bug and all you guys :)

atommann, 2021-03-11:
> 
> 
> Different ways to work around the CAM software problem.
> 
> Does Kicad uses G36/G37 only for octagons or for all pads ?
> 
> And for information, it seems that oblong is not a problem for CAM software.

You can check the Gerbers in this post: [Test files from KiCad](https://forum.kicad.info/t/apologies-from-atommann-an-engineer-from-jlcpcb-about-gerber-x2-files-at-jlcpcb/26847/43?u=atommann)

dbrgn, 2021-03-11:
Thanks a lot for your investigation!

So the problem is a wrong operator precedence, right?

Are the tokens evaluated right-to-left, or is the precedence simply wrong (- before *)? (In the gerbv thread, the evaluation order seems to be left-to-right, which doesn't seem to be what your CAM software does.)

atommann, 2021-03-11:
> Thanks a lot for your investigation!
>
> So the problem is a wrong operator precedence, right?
>
> Are the tokens evaluated right-to-left, or is the precedence simply wrong (- before *)? (In the gerbv [...]

For CAM tool A, according to my test and probing, expression ```A/B - C``` is evalulated as ```A/(B-C)```.

dbrgn, 2021-03-11:
I wonder what `A/B-C/D` would evaluate to. Is it `A/(B-C)/D` or `A/(B-(C/D))`?

ubruhin, 2021-03-11:
> I wonder what A/B-C/D would evaluate to. Is it A/(B-C)/D or A/(B-(C/D))?

Does that matter? The CAM software is buggy, so we should not rely on operator precedence of macros at all :wink:

Anyway, great job @atommann, thanks for investigating this issue! I think for LibrePCB users it's very helpful if we can improve the compatibility of our Gerber files with your CAM software.

After all, I think we should keep using aperture macros, but only with hardcoded numbers to keep the risk of incompatibilities with CAM software low. Probably this will not lead to much larger Gerber files. Btw, I just saw a Gerber file generated by Altium, which used macros with hardcoded values as well (however, I didn't investigate this in detail, I just took a look at a single Gerber file).

When not using aperture macros at all, there are two options: regions (like planes) and stroke fills. But Mr. Tavernier from Ucamco stated *very* clearly to NOT DO THIS. :smile: Since he's *the* expert regarding this topic, I think it makes sense to follow his recommendation.

I'll pick up this issue and open a PR once I have a suggestion ready how to fix this.

doom-fr, 2021-03-11:
So the CAM tool seems to ignore mathematical priority rules (and start from the end ?)
Regarding what's done by Kicad, I think the solution is to do only one operation at a time and create as many intermediate variable as needed.

dbrgn, 2021-03-11:
> Does that matter? The CAM software is buggy, so IMHO we should not rely on operator precedence of macros at all wink

If software would be either correct, or always evaluate from right-to-left, we could try to group the values in a way that both variants of parsing would yield the same result.

But inlining all numbers seems better, and is probably the right thing to do in most cases. The KiCad variant, where intermediate variables are used, will probably only help in cases where the same value is re-used many times.

closed 2021-04-24 by ubruhin