LibrePCB/LibrePCB #890: Error starting application on Ubuntu 20.04 Wayland

vkk-steigend, 2021-05-26:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.5
Git Revision:
Qt Version:  5.9.5
Operating System: Ubuntu 20.04.2
```

##### SUMMARY
LibrePCB does not open after installing in Ubuntu with Wayland session.

##### STEPS TO REPRODUCE
Install LibrePCB using snap
```
$ sudo snap install librepcb
```
Open LibrePCB
```
$ librepcb
```

##### EXPECTED RESULTS
LibrePCB opens.

##### ACTUAL RESULTS
The following error message appears:
```
This application failed to start because it could not find or load the Qt platform plugin "wayland-egl"
in "".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)
```

##### ADDITIONAL INFORMATION
Tried with installing libqt5gui5-gles, same result.

Workaround:
```
$ librepcb -platform xcb
```

ubruhin, 2021-05-30:
I wonder if this is a problem of LibrePCB itself, or only a Snap package issue :thinking: Could you maybe check if the [AppImage](https://docs.librepcb.org/#_portable_package) has the same issue?

In addition, maybe the package `qtwayland5` or `libqt5waylandclient5` needs to be installed?

ubruhin, 2021-08-20:
Fixed in [librepcb-snap](https://github.com/LibrePCB/librepcb-snap), should now be working properly after updating to the latest snap 0.1.5-2.

closed 2021-08-20 by ubruhin