LibrePCB/LibrePCB #891: DXF Import?

jabadabado, 2021-05-29:
Really Great software!

I though when checking it...the most important is here! But then I could not find any way if importing a PCB shape with reference holes via a DXF file etc. Maybe some other way? Through a picture, PDF, or anything that has the exact sizes to begin the board outline and mounting holes etc.

Did I miss something?

ubruhin, 2021-05-30:
Hi,

Currently there is no way to import shapes or drills. Either you have to draw them manually, or for very complex shapes it might be worth to create some script to convert DXF to LibrePCB polygons (e.g. using [`librepcb-parts-generator`](https://github.com/LibrePCB/librepcb-parts-generator) which is able to generate such polygons) and then copy them into the `board.lp` file :wink:

But for long term, of course it makes sense to add this feature to the board editor.

dryas, 2021-07-06:
I just want to add that this is the feature I miss most at the moment. Highly appreciate if we can get some solution here. Thanks for all the great work so far! :-)

EdizonTN, 2021-07-29:
Please, can you add this feature to the library package editor also? 
Will be great way, how to create complicated footprints and place pads to the right positions (in cooperation with snap function).

PR #917 (2021-09-24): Add DXF import to package-, symbol- and board editors
## Summary

This adds a DXF import to the package editor, symbol editor and board editor. This is especially useful to import board outlines or package dimensions from a mechanical CAD drawing, to avoid the need to manually draw these polygons.

(Although the symbol editor is not really something mechanical, it might still be useful to import a logo or so.)

Closes #891

## Dialog

The dialog allows to configure how to import the DXF:

- Layer of the imported lines
- Optionally import circles as drills instead of polylines (e.g. for board m [... cut]
closed 2021-09-24 by ubruhin