LibrePCB/LibrePCB #893: LibrePCB library conversion v0.1→v0.2 crashes on macOS

dbrgn, 2021-06-07:
When testing the cmake builds on macOS 11.1 (Big Sur), I noticed that LibrePCB crashes with my v0.1 library when trying to convert it. It's not cmake specific though, the same thing happens with the nightly builds from `master`.

This may be due to a corrupt database, but even then, nothing should crash. To me, it looks like a bug related to the file system.

##### DETAILS

```
Crashed Thread:        11  librepcb::AsyncCopyOperation

Exception Type:        EXC_ARITHMETIC (SIGFPE)
Exception Codes:       EXC_I386_DIV (divide by zero)
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Floating point exception: 8
Termination Reason:    Namespace SIGNAL, Code 0x8
Terminating Process:   exc handler [937]

Thread 11 Crashed:: librepcb::AsyncCopyOperation
0   org.librepcb.LibrePCB           0x0000000102a0fff2 librepcb::AsyncCopyOperation::run() + 658
1   org.qt-project.QtCore           0x0000000103ccfb29 0x103cae000 + 138025
2   libsystem_pthread.dylib         0x00007fff20383950 _pthread_start + 224
3   libsystem_pthread.dylib         0x00007fff2037f47b thread_start + 15
```

Full crashlog: [crashlog.txt](https://github.com/LibrePCB/LibrePCB/files/6611612/crashlog.txt)

##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     a435c5095c6f94df4da130fe832ea97f42d17901
Build Date:       2020-10-03T19:49:28
Linking type:     static
Unbundled libs:   -
Qt Version:       5.15.1 (built against 5.15.1)
CPU Architecture: x86_64
Operating System: macOS 10.16
Platform Plugin:  cocoa 
```

DMG: https://tmp.dbrgn.ch/librepcb-nightly-mac-x86_64.dmg
Database v0.1: https://tmp.dbrgn.ch/v0.1.tar.gz

##### STEPS TO REPRODUCE

1. Put v0.1 directory in ~/LibrePCB-Workspace/
2. Open LibrePCB
3. When being asked to import the library from v0.1 format (dialog "Initialize Workspace"), accept
4. In the next dialog, confirm with "Start Import"
5. LibrePCB crashes

ubruhin, 2021-06-19:
Oooh very bad bug :scream:

Thanks for reporting so detailed, it helped to easily reproduce it :muscle: -> fixed in #896.

Removed the macOS label since it is not related to macOS.

dbrgn, 2021-06-19:
> fixed in #896

Cool!

PR #896 (2021-06-20): Fix potential errors or crash when migrating workspace file format
Due to several issues in `AsyncCopyOperation`, migrating a workspace from v0.1 to v0.2 might have failed or even crashed in some cases -> fixed these issues and added corresponding unit tests.

Fixes #893.
closed 2021-06-20 by ubruhin