LibrePCB/LibrePCB #894: macOS Installer: Wrong application bundle format

dbrgn, 2021-06-07:
With the current installer (both 0.1.5 stable and the version built by the `cmake` branch), the installer does not use the proper application bundle format for macOS.

This is how it looks:

[image]

I'm not really familiar with macOS packaging, but this should not be a directory containing shared files and a binary, but instead it should be a self-contained application bundle that can be double-clicked directly. It works, but it's not nice.

The drag&drop dmg app bundle on the other hand works properly and shows a "librepcb"-entry with icon in the applications directory.

ubruhin, 2022-10-02:
Finally I was able to extensively test LibrePCB on macOS and I also saw this issue.

I'd say there's nothing wrong with the bundle format. The installer itself creates some configuration files in its installation directory (we can't install just the plain bundles), therefore we shouldn't install our files directly to `/Applications`.

Probably the main issue is that we set `/Applications` as the default installation location (which we did because of #644). Actually the files should be installed somewhere else (in a normal data directory), and the installer should just create symlinks in `/Applications` for the relevant applications (actually only for LibrePCB itselt, since the CLI makes no sense to be opened through the GUI). On Linux, a typical installation directory would be `/opt` or `/usr/local`, but I don't know if there's a corresponding directory on macOS.

In the end, maybe we should simply consider not providing an installer for macOS at all. It seems that plain *.dmg bundles are the way to go on macOS. But then probably we should provide a built-in updater to allow the user to easily update LibrePCB.

Actually I even wonder if this would make sense for Windows and Linux as well - maintaining the installer is a bit cumbersome anyway, so I'd be quite happy to completely get rid of it :see_no_evil: But at the moment it is the only convenient way for users to keep LibrePCB up to date...

dbrgn, 2022-10-02:
In my opinion installers are only a thing on Windows. It feels wrong to use an installer on Linux, and from my limited experience, DMGs are the way to go for macOS.

PR #1147 (2023-04-11): Stop providing installers for Linux and MacOS
This is not an easy decision, but given the facts that:

- Installers are quite unusual on Linux and MacOS
- The installer on MacOS creates quite a mess in the launcher (#894)
- The AppImage is now able to install mime types and *.desktop files (#1146)
- There are plenty of other installation methods on Linux (Snap, Flatpak, distribution packages, ...)

I think it makes no longer sense to maintain the installers for Linux and MacOS. The main drawback is that with the portable packages we don't have an update mechanism (while the installers *do*) bu [... cut]
closed 2023-04-11 by ubruhin