LibrePCB/LibrePCB #901: Board Editor : artefact in one rounded shape

doom-fr, 2021-07-21:
##### VERSION / OS / ENVIRONMENT

It is the current nightly build branch 0.1

```
LibrePCB Version: 0.1.5
Git Revision:     20f2c30ac403cc69fa0db4cbb85acfb43681abc0
Build Date:       2021-07-11T12:00:30
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY
There's an artefact in the drawing of the curve
[image]

##### STEPS TO REPRODUCE
??

##### EXPECTED RESULTS
The curve is smooth
[image]
(image from another segment)

##### ACTUAL RESULTS
There's an artefact

##### ADDITIONAL INFORMATION
[image]

The problem occurs with -10° angle at line 27.
Changing angle (less than 0) doesn't fix anything.
Moving the point doesn't fix anything
Replacing the point by another one doesn't fix the problem too.
Setting 10° solve, but it is not what's needed.

doom-fr, 2021-07-21:
Here is a minimal example with less points : 

[image]

doom-fr, 2021-07-21:
Is it only a displaying problem or there will be a problem with gerber files ?

ubruhin, 2021-07-21:
I suspect this is only a rendering issue, but you can easily check the Gerber output with a Gerber viewer.

Anyway, if the deviation is extremely small, I think it should not lead to any problems, even if still contained in the Gerber output.

doom-fr, 2021-07-22:
It seems it is not on gerber files.