LibrePCB/LibrePCB #902: Board Editor : Make Planes less visible

doom-fr, 2021-07-21:
##### SUMMARY
When you have 4 layers board and display the planes, it really difficult to see anything.
It is needed to show/hide planes and also each layer to know where planes are or not.

##### SOLUTION
I think it will be preferable to display shaded area : 
[image]

[image]

It will be also great to have a keyboard shortcut to show hide the plane.

doom-fr, 2021-07-21:
I think in Kicad there's both display modes.

ubruhin, 2021-07-21:
You know the "Visible" checkbox in the context menu of planes, and the global show/hide all planes menu item?

However, I agree there might be better options than just show/hide the planes.

doom-fr, 2021-07-21:
Yes I know obviously show/hide all planes.
But no, do not know "Visible" in context menu. Thanks for pointing me.
I will test it but I'm not sure it will help a lot.