LibrePCB/LibrePCB #904: Board Editor : Impossible to activate the Via tool

doom-fr, 2021-08-18:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.5
Git Revision:     20f2c30ac403cc69fa0db4cbb85acfb43681abc0
Build Date:       2021-07-11T12:00:30
Linking type:     static
Unbundled libs:   -
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
```

##### SUMMARY
Error "No Net Signal Selected" and impossible to activate the via tool.

##### STEPS TO REPRODUCE

##### EXPECTED RESULTS

##### ACTUAL RESULTS

##### ADDITIONAL INFORMATION
Need to close/open the project

ubruhin, 2021-08-19:
This error appears if there are no net signal existing at all. Does your schematic contain any nets when this error occurs? If it does, could you provide the exact steps how to reproduce this issue?

ubruhin, 2022-03-12:
This is fixed in #854 since it allows to create vias without electrical connection. However, since it required a file format change, it will not be available in 0.1.x releases, but in 0.2.0.

closed 2022-03-12 by ubruhin