LibrePCB/LibrePCB #913: Illumos: Qt signals don't work

dbrgn, 2021-09-15:
When building LibrePCB on Illumos (OpenIndiana):

```
$ mkdir build && cd build
$ CMAKE_PREFIX_PATH=/usr/lib/qt/5.12/ cmake .. -DCMAKE_BUILD_TYPE=Debug
$ make -j 20
```

[image]

...and running the resulting binary, the signals don't seem to work.

[image]

"QObject::connect: signal not found in QPushButton (:0)"

Even though the text field is filled out, the button is still greyed out. See also the highlighted warnings on the left.

This did not happen with the old qmake based setup, so it may be related to some build system details in combination with Illumos 😕 LTO is not being used, as far as I can tell.

(Maybe @Toasterson knows something?)

Toasterson, 2021-09-16:
We had some events not being passed to glib/HAL some time ago but only USB connect events. Keystrokes worked always for other Languages... And I am not aware of other QT applications having problems. 

Is the build system skipping this because of some missing library or a header type we have differently than other OSes? I know we get Problems in other C++ apps because our int8_t is implicit signed char rather than explicitly signed char

ubruhin, 2022-09-18:
I tried to reproduce this with the latest OpenIndiana release. Everything seems to work, no problem with signals&slots anymore! :rocket: 

[image]

Probably the issue was not on our side, so let's close it.

closed 2022-09-18 by ubruhin