LibrePCB/LibrePCB #916: Having a lot of problems getting this to build on Windows 10

masondb, 2021-09-21:
I've installed Qt 5.12 on a new laptop with Windows 10 and tried to build everything and was running into issues with Zlib. I uninstalled Qt and installed Qt 5.15 and it ran into no issues but now when I try to build in Qt Creator, I run into issues with "Qt core: No file found in directory" in muparser. I'm using Mingw in the newest version of Qt Creator 5.0.1. 

I've been trying to get this figured out for almost two days at this point so I can get started contributing to the project but Qt keeps breaking. I've never had problems with building any other software before. I spent hours on google only to find nothing. The include directory points to the correct include folder for Qt's headers.

Has anyone actually had this able to build properly on Windows 10 without much issue?  Unfortunately, I've run into issues with several distros of Linux being incompatible or unstable with my hardware so that isn't an option or I would have used it now. 

Exact errors:

C:\Users\MB\Documents\Git\LibrePCB\libs\muparser\include\muParserDef.h:32: error: QtCore: No such file or directory
In file included from C:/Users/MB/Documents/Git/LibrePCB/libs/muparser/include/muParserCallback.h:33,
                 from C:/Users/MB/Documents/Git/LibrePCB/libs/muparser/src/muParserCallback.cpp:29:
C:/Users/MB/Documents/Git/LibrePCB/libs/muparser/include/muParserDef.h:32:10: fatal error: QtCore: No such file or directory
 #include <QtCore>
          ^~~~~~~~

masondb, 2021-09-21:
Deleting the local copy and cloning it again then opening it again in Qt Creator seemed to let it build fine this time with Cmake...weird but working with the CMake copy kinda sucks. Is there any easy way to generate a Qtcreator compatible project directory that actually lists the .ui files in the project like a regular Qt project?

masondb, 2021-09-21:
Got everything working now, Qt Creator wasn't cooperating but it's all good.

closed 2021-09-21 by masondb

ubruhin, 2021-09-21:
Hm I never had such issues with qtcreator, so far it always worked fine - both with qmake and with cmake.

If you still have issues, it would be good to see the complete console output when compiling. The error message "muParserDef.h:32:10: fatal error: QtCore: No such file or directory" makes no sense at all since muParser does not include any Qt files :wink:

masondb, 2021-09-21:
> Hm I never had such issues with qtcreator, so far it always worked fine - both with qmake and with cmake.
>
> If you still have issues, it would be good to see the complete console output when [...]

Yeah, I noticed that. I think maybe something got messed up in the process, but doing a clean build in Qt Creator after installing Qt 5.15 fixed my issue and everything works fine now. Hopefully I can start contributing soon. 

Thanks for taking the time to reply.