LibrePCB/LibrePCB #919: SSL errors occurred: The certificate has expired

Symbian9, 2021-10-01:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.6-rc1
Git Revision:     32a1097
Build Date:       2021-09-30T11:36:43
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Debian GNU/Linux 10 (buster)
Platform Plugin:  xcb
```

##### SUMMARY

LibrePCB can't load libraries via "Library Manager"

##### STEPS TO REPRODUCE

1. Launch LibrePCB;
2. Open "Library Manager"
3. See error message.

##### EXPECTED RESULTS

"Library Manger" should work without issues.

##### ACTUAL RESULTS

"Library Manager" does not work due to next issue:

> SSL errors occurred: The certificate has expired

##### ADDITIONAL INFORMATION

ubruhin, 2021-10-01:
That's very bad :sob: But I don't understand yet why this happens, actually the certificate of api.librepcb.org looks valid. Maybe it's somehow related to the OpenSSL library bundled within the AppImage :thinking: At least I can't reproduce this problem when building LibrePCB from sources.

Symbian9, 2021-10-01:
> Maybe it's somehow related to the OpenSSL library bundled within the AppImage

Take a look on next issues:

- https://github.com/antoniofaccioli/libreoffice-appimage/issues/15
- https://github.com/openssl/openssl/issues/7481

Symbian9, 2021-10-01:
> At least I can't reproduce this problem when building LibrePCB from sources.

What Linux distro you use? I use Debian, and there are some default path may be differ from Ubuntu or other dstros.

ubruhin, 2021-10-01:
I think I have found the reason for this problem. Older OpenSSL library versions do no longer work with recent Let's Encrypt certificates: https://community.letsencrypt.org/t/openssl-client-compatibility-changes-for-let-s-encrypt-certificates/143816

There seem to be a server-side workaround available, but I do not understand yet how to implement this workaround exactly. I'm working on it... In the worst case (if the workaround doesn't work) I think we have to create a new LibrePCB release.

> As side note, does OpenSSL really needed to be included in AppImage? Why just not left its outside and lets LibrePCB-*.AppImage would use installed openssl package on each distro instead?

That's a good question. Of course this avoids the dependency to OpenSSL, but as we see now, bundling also leads to problems :see_no_evil: 

Btw, our Windows builds are still working so probably only our Linux binaries (any maybe macOS) are affected by this issue.

Symbian9, 2021-10-01:
> Btw, our Windows builds are still working so probably only our Linux binaries

Does Windows builds also include same OpenSSL version as it bundled into AppImage?

ubruhin, 2021-10-01:
I think I could fix it server side! :rocket: @Symbian9 can you confirm it is working on your setup as well?

However, it's just a temporary workaround. Now we should think about how to avoid such issues in the future.

Symbian9, 2021-10-01:
> can you confirm it is working on your setup as well?

Confirmed!

[image]

Symbian9, 2021-10-01:
> Now we should think about how to avoid such issues in the future.

For any AppImage specific help:

- https://github.com/AppImage/AppImageKit/wiki#-where-do-i-get-support

dbrgn, 2021-10-03:
> I think I could fix it server side!

What did you change?

ubruhin, 2021-10-03:
> What did you change?

I switched to the alternate certificate chain as explained in https://community.letsencrypt.org/t/openssl-client-compatibility-changes-for-let-s-encrypt-certificates/143816.

ubruhin, 2021-10-12:
> This is only for our AppImage binary releases, right?

No, it's for all binaries. Windows, Linux, macOS; Installers, archives and AppImages.

> Keep in mind: This means we should probably publish new builds if there are any critical OpenSSL vulnerabilities being published (affecting the version we are bundling).

That's what I meant with "[...] and try to keep it up to date in future.". However, I don't think we have to create new LibrePCB releases each time a new vulnerability is known since LibrePCB is not a security critical application like a web browser. The API is accessed only rarely and supports only a very limited set of features. Also the downloaded libraries are not confidential at all (thus I thought about even using HTTP instead of HTTPS). So IMHO it would be totally fine to just update OpenSSL when we create a new release anyway. Yes, this opinion might be different to yours :wink: