LibrePCB/LibrePCB #933: cmake infinite recursion when finding gtest and gmock

MorganJamesSmith, 2021-11-08:
I get this error when building. I'm using the compile flag -DUNBUNDLE_GTEST=ON.

I don't understand how cmake works but my guess is line 24 in cmake/FindGTest.cmake where it tried to find_package(Gtest) is actually running the FindGTest.cmake file again. 

```
CMake Error at cmake/FindGTest.cmake:1 (set):
  Maximum recursion depth of 1000 exceeded
Call Stack (most recent call first):
  cmake/FindGTest.cmake:24 (find_package)
  cmake/FindGTest.cmake:24 (find_package)
  cmake/FindGTest.cmake:24 (find_package)
  cmake/FindGTest.cmake:24 (find_package)
  ...
  (lots more of the same here)
  ...
  cmake/FindGTest.cmake:24 (find_package)
  CMakeLists.txt:141 (find_package)
```

ubruhin, 2021-11-11:
@dbrgn any idea regarding this issue?

dbrgn, 2021-11-11:
Hi, I had this issue as well when packaging on Arch Linux 😕 During development this worked, so I'm not sure why.

> I don't understand how cmake works but my guess is line 24 in cmake/FindGTest.cmake where it tried to find_package(Gtest) is actually running the FindGTest.cmake file again.

Yeah, I think it's something like that. Probably has to do with how the CMake files bundled with GTest are written.

PR #1013 (2022-09-24): CMake: Fix & improve unbundling of GTest/GMock
Adding `CONFIG` to the `find_package()` seems to fix the endless recursion which was reported in #933. I don't understand what `CONFIG` means exactly, but at least [in this](https://chromium.googlesource.com/external/github.com/pwnall/googletest/+/HEAD/googletest/README.md) documentation they also do it this way. And local testing shows that this seems to fix the problem.

In addition, I adjusted `FindGTest.cmake` to proceed even if `find_package(GMock)` failed - `find_package(GTest)` should be sufficient since GTest usually also contains GMock, s [... cut]
closed 2022-09-24 by ubruhin