LibrePCB/LibrePCB #934: Board Editor: Add/Edit PCB Layers

EdizonTN, 2021-11-12:
Hi all,

I tryied to compare a few EDA system with LibrePCB from layers point of view. This will be good to add a LibrePCB manual or use as import/export layer mapper.
[Here is](https://docs.google.com/spreadsheets/d/1RAiLkSRvMO-v-Gyn5_CpmHMhc6j0mbS5gSQYTTUWYaQ/edit?usp=sharing).
btw, if is some KiCAD user here, I will glad if check KiCAD layers in this document. Thanks

I found, that some layers will be good to add or change to LibrePCB:
1. Sheet Frames layer - should be divided to Top and Bottom, because print of the PCB from bottom view (mirrored) needs also frame with title block opposite to top view, which is not mirrored view

2. Milling (PTH) layer - should be dividied to two layers: Milling (PTH) and Milling (NPTH). Yes, milling can be also NPTH
Or change to "Milling" only and PTH vs NPTH attribute can be add to milling lines properties

3. Drills NPTH layer - hmm, same as milling. Divide to PTH and NPTH, or describe it with attributes.

4. Airwires layer - really is good idea to add as layer? For example (in the future), you will need to change color for specific airwires (VCC, GND,....) - will be possible in the one layer?

5. Measures layer - should be divided to Top and Bottom, because some drawed dimensions need be placed from top layer of the PCB, but from bottom of the PCB also! And from bottom view - so mirrored.

6. Alignment layer - I don' understand of function of this layer

7. Add a user's layers (mechanical) - this is important ! With possibility to paired (as symmetric) two together.

Thanks
Ed

ubruhin, 2021-11-12:
Hi,

I agree layers are not perfect so far - it is quite complex to find a good system (especially for special functions like airwires, as you already figured out). However, here some clarifications regarding your comments:

> Milling (PTH) layer - should be dividied to two layers: Milling (PTH) and Milling (NPTH). Yes, milling can be also NPTH
Or change to "Milling" only and PTH vs NPTH attribute can be add to milling lines properties

The idea is that NPTH millings should be drawn on the board outline layer, since AFAIK no separate Gerber file is needed for NPTH millings. It's just the same as the normal board outlines. The milling layer exists only to explicitly create PTH millings.

> Drills NPTH layer - hmm, same as milling. Divide to PTH and NPTH, or describe it with attributes.

Similar idea here - PTH drills is the same as vias or THT pads, thus no separate layer needed. If you need a PTH drill, just add a via instead. However, unconnected vias are not supported yet in v0.1.x, but it will be possible in v0.2.x.

> Airwires layer - really is good idea to add as layer? For example (in the future), you will need to change color for specific airwires (VCC, GND,....) - will be possible in the one layer?

Unfortunately I don't have an answer for this topic yet - I hope I'll find a good solution for this some day :see_no_evil: 

> Alignment layer - I don' understand of function of this layer

It's intended to add lines to footprints, which will help to align the footprint in a board. For example, an edge connector or switch could draw a line where the board outline or panel front side needs to be located to have the device properly aligned. Example (orange line shows position of panel front side):

[image]

EdizonTN, 2021-11-13:
> Hi,
> 
> I agree layers are not perfect so far - it is quite complex to find a good system (especially for special functions like airwires, as you already figured out). However, here some clarifications regarding your comments:

Thank for reading and replying!

> The idea is that NPTH millings should be drawn on the board outline layer, since AFAIK no separate Gerber file is needed for NPTH millings. It's just the same as the normal board outlines. The [...]

I'm sorry, but I think, this is not good solution. Board outline should be only for real board outline. It's maximal area of your board for PCB manufacturer. Without any holes and milled openings.
Board outline is also need for last step in the production - final dimension check (if dimension are precious) or PCBs division from panel. Example: PCB included milling openings but final divide is only by V-CUT. This needs two separate data files (I know, this is a extreme situation, but is possible). 
btw, during generating NORMAL PCB, is no problem put to one file all these layers.

Due this I prefer separate file for each process.
Drill - PTH and NPTH = two files.
Milling - PTH and NPTH = two files
Board outline = separate file.

> Similar idea here - PTH drills is the same as vias or THT pads, thus no separate layer needed. If you need a PTH drill, just add a via instead. However, unconnected vias are not supported yet in [...]
Same as previous :-)
During PCB design process is better divide the mechanical holes (PTH and NPTH) and Pads/Vias separately.
Mechanical holes and openings are obviously fixed and during design, designer can't move their. Opposite of this, vias and pads are moveble during design process (one of design basis).
Will be better, if mechanical holes and vias/pads will be in separate layers. And if mechanical holes (layer) can be locked (important thing).
During CAM generation, can be stored all PTH hole in one file and all NPTH in one file. This can be selectable - which layer go to PTH file.

> Unfortunately I don't have an answer for this topic yet - I hope I'll find a good solution for this some day 🙈
OK, if LibrePCB will have option select color for each airwire in the future - can stay as is.

> > Alignment layer - I don' understand of function of this layer
> 
> It's intended to add lines to footprints, which will help to align the footprint in a board. For example, an edge connector or switch could draw a line where the board outline or panel front side needs to be located to have the device properly aligned. Example (orange line shows position of panel front side):
> 
> [image]
> 
> After you aligned the device, you can hide this layer since you don't need it anymore.

Thanks for explanation!
This is a typical user's layer from my point of view. I use this alignment lines also, but I go litlle bit on.  I used (in my previous EDA system) also one user's layer for connector insertion.
Ex.: I need place a two PCB mounted connector to the PCB. How close are can be one to one? What should be a minimal manipulation space?
I switch on my special layer, with drew a opposite connector (cable end) and I see a minimal space for cable insertion.
Due this, My suggestion is:
Don't create a special layer for alignment. Create a more user's definable layers, and alignment use only as suggestion (in manual).
Of course, add to each user's layer a option, for pairing with another user's layer (element can be switched from top to bottom and opposite).

Next questions appearing - which layer come from library parts? Which layers can get imported objects (DXF,...)? Which layers are electrical, mechanical or graphical only (ERC, DRC, no check) and which can be paired? Which layers can be locked by user?
Due this questions, should be user's layers fully definable, I think.

p.s. Have you any general proposal for LibrePCB's layers? Will be good, if more PCB designer can comment this from their practice. Maybe at another place (discourse.group?).

Thank you!
Ed.

ubruhin, 2025-08-17:
I'd like to get this resolved for LibrePCB 2.0 so I thought about it again...

Sheet frames: Just splitting this layer into separate top/bottom layers doesn't fix the problem. Currently board sheet frames don't work at all because the scaling is fixed. For example when we add an A4 sheet frame to a board which is only 10x10mm, the board would be printed way too small on the paper. What we really want is to either fit the board into the sheet frame, or apply a manual scale factor to the board. We first need to define how to handle board sheet frames at all before we can decide what layers we need. I'm currently not sure what is a reasonable way to go...

Milling: Already fixed in v1.x as there are now three separate layers (board outline, board cutouts, plated board cutouts).

Drills: Still the same opinion as above - I think we don't need a plated drills layer because those are in fact vias. Might be re-considered when implementing #1078.

Measures: Not sure about this. Somehow I agree split top/bottom layers would make sense. But somehow I never understood what measurings of boards are useful for. Why are people adding measurements to boards? I'm not sure. And I even disagree to send such measurings to PCB manufacturers. All values are in the Gerber files, which is the single source of truth.

User-defined layers: Here I'd be interested in real-world use-cases. In my experience, user-defined layers break the portability of libraries and projects of other EDA tools. I always got angry when I received a library or project which contained nonstandard layers as they are often annoying, not self-explaining what to do with them, leading to export issues, or even conflicting with own custom layers. So I didn't want to add them in LibrePCB without a very very good reason. And so far, except from this thread, I've never got a request to add user-defined layers. So I am waiting for real-world use-cases before I will think about it.

Mauro-Design-PCB, 2026-05-16:
Thanks for the clear explanations, however, the important point is that system layers must be kept separate from user layers. User layers should only be used for special cases that frequently arise in design, while system layers should always be used in other cases. For example, special cases include those where two or more designers are sharing the project development and their annotations are therefore personal. It's a good idea to incorporate their names into the layer names so other designers can identify who made them.

It's therefore very important that the EDA system supports user layers not only in the PCB editor but also in the schematic editor, even before printing, by enabling or disabling their display during the work session. Depending on the layer visibility settings in the editors, the subsequent printing process will only include the currently visible layers.

I'd like to clarify that the ability to create, delete, and edit additional user layers, as well as toggle their visibility, is an important resource in EDA system editors, especially considering that they are 2D CAD systems. They provide the flexibility needed to support all future needs that may arise during design.

I'd also like to clarify that user layers do not cause confusion or conflict with system layers because they are managed separately and are, in fact, an indispensable resource in many situations. However, they should never be used to replace system layers, nor should they force the layer change of elements already inserted on system layers.

I hope I've clarified the reasons why an EDA system, like a 2D CAD system, must have the ability to insert user layers, and that all schematic and PCB editors also have the ability to toggle their display for system layers, for maximum operational flexibility.

To realize the extreme usefulness of user layers, I invite you to install the latest version of Cadsoft's Eagle 7.7.0 software (found on an online repository) in free mode (Eagle Express License) and try the Layer management functionality (system and user) with the display (layer settings) command.