LibrePCB/LibrePCB #935: 0.1.6 installs an empty directory share/librepcb/i18n

yurivict, 2021-11-19:


ubruhin, 2021-11-19:
I can not reproduce this issue - our official releases do contain the correct i18n files.

Can you please provide some more information? What commands did you execute to clone and build LibrePCB? Any console output?

Does the source directory `i18n` contain \*.ts files? If not, probably the wrong commit of the `i18n` Git submodule is checked out.

yurivict, 2021-11-19:
> Does the source directory i18n contain *.ts files?

No. It is cloned from [this recision](https://github.com/LibrePCB/librepcb-i18n/tree/efc96aa73efe8a1410e8c7f4d256713e67524ae5) of the repository that doesn't contain *.ts files.

ubruhin, 2021-11-19:
OK so then you probably just need to execute `git submodule update --init --recursive` (as described in our README.md) after checking out the 0.1.6 tag. Then the submodule should point to a commit which does contain *.ts file.

yurivict, 2021-11-19:
I switched the FreeBSD port to the tarball which doesn't have such problem.

Thank you for your help.

closed 2021-11-19 by yurivict