LibrePCB/LibrePCB #942: License on Icons8 images

FedericoCeratto, 2021-12-25:
Hi. Can you please help me clarify the licensing status of the icons from Icon8 listed in https://github.com/LibrePCB/LibrePCB/blob/master/img/README.md?plain=1#L84 

It appears the license for free accounts does not allow derivative work and redistribution making it not FOSS: 

From https://intercom.help/icons8-7fb7577e8170/en/articles/5534926-universal-multimedia-licensing-agreement-for-icons8

`
Free Account Holders must not make derivative copies of any icons or music accessed, used, or downloaded from our website; however, Free Account Holders may use Icons8 Desktop Tools (Photo Creator, Ouch Creator, Icon Effects, OMGIMG, Pichon, Lunacy) to modify or change illustrations or photographs.
`

ubruhin, 2021-12-26:
Hi,

hmm yeah I think most or all of the icons we use from Icons8 are published under the so-called "linksware license" and is therefore not FOSS. As far as I know, we are allowed to use their images since in the about dialog we contain a link to Icons8 and we didn't modify their icons.

One interesting statement on their license page sounds good:

> Established open-source projects could receive the icons for free. For you, we'll open-source the needed resources.

But I didn't find any further information, what they exactly mean with "open-source the needed resources".

@dbrgn do you know more about the Icons8 license situation? Otherwise we could ask for more information in their [support forum](https://community.icons8.com/)...

dbrgn, 2021-12-26:
> @dbrgn do you know more about the Icons8 license situation?

No, as far as I can remember I thought the license was much simpler than [the link posted by @FedericoCeratto](https://intercom.help/icons8-7fb7577e8170/en/articles/5534926-universal-multimedia-licensing-agreement-for-icons8)... Maybe it was updated in the meantime.

FedericoCeratto, 2021-12-26:
@ubruhin I asked and they pointed me to https://icons8.com/license
`
Established open-source projects could receive the icons for free. For you, we'll open-source the needed resources.
By established projects, we mean the Wikipedia criteria. If you have a page on Wikipedia, and it's not removed within a week, you're qualified.
`

ubruhin, 2021-12-27:
> I asked and they pointed me to https://icons8.com/license

Hmm okay, whatever they mean exactly with that sentence...

Btw I just opened [a PR](https://github.com/LibrePCB/librepcb-website/pull/54) to put our website content under CC-BY license to allow copying content to Wikipedia (see [this issue](https://github.com/LibrePCB/librepcb-website/issues/42)). Hopefully this brings us one step closer to having a page on Wikipedia, which would allow us to take advantage of that Icons8 offer.

FedericoCeratto, 2021-12-27:
@ubruhin the offer for "free use" might not be under a real open-source license, e.g. have limited redistribution and modification rights.