LibrePCB/LibrePCB #943: Board Editor : Bug : can't select some polygons of board outlines

doom-fr, 2021-12-26:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.6
Git Revision:     c4f3164
Build Date:       2021-11-02T19:29:07
Qt Version:       5.15.2 (built against 5.15.2)
CPU Architecture: x86_64
Operating System: Ubuntu 20.10
Platform Plugin:  xcb
TLS Library:      OpenSSL 1.1.1l  24 Aug 2021
```

##### SUMMARY
Some (but not all) polygons (of board outlines) can't be selected, modified, etc ...

##### STEPS TO REPRODUCE
I just opened an 0.1.5 project in 0.1.6
I see that some polygons can't be edited

##### EXPECTED RESULTS
When I double-click, properties window of polygon should open
When I right-click, I should see the contextual menu
When I select rectangle, nodes should be focused and drawn

##### ACTUAL RESULTS
When I double-click => nothing
When I right-click => nothing
When I select rectangle => nothing

##### ADDITIONAL INFORMATION

doom-fr, 2021-12-27:
:sweat_smile: It seems not a bug in fact.
The polygon is in fact in a footprint.

So can I transform this into a request to just tell it was in a footprint ?

doom-fr, 2021-12-27:
Maybe it's not relevant.
Maybe it's better to just close this for now.

ubruhin, 2021-12-27:
OK no problem, then I'll close it :)

closed 2021-12-27 by ubruhin