LibrePCB/LibrePCB #944: UI Blurry on macOS with HiDPI w/ quick fix

ClassicOldSong, 2021-12-28:
##### SUMMARY

UI and text blurres very much on macOS with HiDPI.

##### SOLUTION

Add the following lines to `Info.plist` should fix the problem:

```
<key>NSPrincipalClass</key>
<string>NSApplication</string>
<key>NSHighResolutionCapable</key>
<string>True</string>
```

Source: https://doc.qt.io/qt-5/highdpi.html

ubruhin, 2022-01-01:
Hi @ClassicOldSong,

thanks for the hint. I added these lines to `Info.plist` on a branch, could you verify that this fixes the problem? Here is the *.dmg with the fix: https://download.librepcb.org/nightly_builds/944-macos-enable-hidpi/librepcb-nightly-mac-x86_64.dmg

It's a nightly build with file format 0.2, so please only use it for opening files but not for saving them.

I just hope it does not break non-HiDPI displays - if you are able to test on lower resolutions as well, that would be helpful too.

ClassicOldSong, 2022-01-02:
@ubruhin Tested and seems fine, it should not break non-HiDPI systems in anyway.

ubruhin, 2022-01-02:
Perfect, thanks for testing!

PR #945 (2022-01-02): MacOS: Enable HiDPI support in Info.plist
Seems to fix the HiDPI issue on macOS, see #944.

Closes #944.
closed 2022-01-02 by ubruhin