LibrePCB/LibrePCB #948: Board Editor : Bug : Round shaped plane is half filled

doom-fr, 2022-01-14:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.6
Git Revision:     c4f3164
Build Date:       2021-11-02T19:29:07
Qt Version:       5.15.2 (built against 5.15.2)
CPU Architecture: x86_64
Operating System: Ubuntu 21.10
Platform Plugin:  xcb
TLS Library:      OpenSSL 1.1.1l  24 Aug 2021
```

##### SUMMARY
Round shaped plane is half filled

[image]

##### STEPS TO REPRODUCE

[image]

##### EXPECTED RESULTS
full circle filled

##### ACTUAL RESULTS
half circle filled

##### ADDITIONAL INFORMATION

This works : 
[image]
[image]

ubruhin, 2022-01-14:
Hm generally the problem with your plane is that the last vertex is missing - a closed path requires at least 3 vertices, not 2. But of course a half-filled plane is not the behavior we expect, even when a vertex is missing. Either we should handle planes like polygons - which are not filled at all if they have less than 3 vertices - or we could automatically add an implicit last vertex to make it a closed path even if the last vertex is missing (because non-closed planes make no sense, in contrast to polygons). I'll try to decide and fix this for the next release.

Thanks for reporting!

doom-fr, 2022-01-15:
I do not know exactly what should be done, but this is mandatory : 
if we see a full circle for the outline of the plane, it should fill the full circle.

PR #961 (2022-03-13): Fix half-filled plane if outline is not a closed path
Non-closed plane outlines (i.e. last vertex position != first vertex position) make no sense so lat's implicitly consider them as closed paths. This way they are filled properly even if the outline is not a closed path.

Fixes #948.
closed 2022-03-13 by ubruhin