LibrePCB/LibrePCB #956: Change transformation order of mirror and rotate?

ubruhin, 2022-03-04:
##### SUMMARY

Currently we apply all coordinate transformations in the following order:

1. Rotate (CCW)
2. Mirror (invert X coordinates)
3. Translate (move)

Such coordinate transformations are used for example to determine the placement of symbols in a schematic, or packages and their pads in a board. Local symbol/package coordinates (relative to the symbol/package origin) are transformed into global coordinates (relative to the schematic/board origin) this way.

Since rotation is applied before mirroring, the rotation direction depends on the mirroring. Normally the rotation is counterclockwise, but if mirroring is active, it is clockwise. As an example, see this package which rotation is set to 10°:

| non-mirrored (top) -> CCW | mirrored (bottom) -> CW
|---|---|
| [image] | [image] |

It sounds trivial and not really relevant. However, it turned out many times to be very cumbersome to implement. Applying coordinate transformations and user commands the correct way is really hard sometimes. And there were already several bugs related to these transformations (e.g. #605, #955, #493, #534).

One reason is that rotation and translation are trivial transformations, while mirroring isn't. For example the Gerber export only provides translation and (sometimes) rotation, but there is no support for mirroring objects. Therefore we have to apply mirroring manually, e.g. by calculating the mirrored result of a polygon. But then the rotation needs to be applied manually as well, because mirroring takes place *after* rotation. Similar problems occurred in other parts of the software, not just the Gerber export. Applying the manual transformation (mirroring) first would be easier since rotation can then be handled as usual by the underlying export feature (Gerber generator, PDF writer, ...).

##### SOLUTION

Therefore I wonder if it would make more sense to change the order of coordinate transformation to apply mirroring first:

1. Mirror (invert X coordinates)
2. Rotate (CCW)
3. Translate (move)

I suspect this would make the implementation much simpler, more efficient and less error-prone. And regarding usability it doesn't make a big difference, maybe it would even be a bit more intuitive since the rotation direction would no longer be inverted for mirrored objects. At least for me it feels more intuitive to apply mirroring first.

I also took a look at Eagle and KiCad how they perform coordinate transformations. Eagle uses the same system as we use now, which also makes it easy to convert Eagle library elements to LibrePCB since no coordinate transformation needs to be applied. KiCad uses the system I proposed here, with the only difference that they are mirroring the other way around (invert Y coordinate, not X). Somehow this mirror direction does not feel intuitive, which might be the reason why their mirror command in the board editor does not only toggle the mirror flag, but also rotates by 180° - which basically leads to mirroring the other way around :sob:

So when we change the coordinate transformation to my proposal, I see the following drawbacks:

- It would no longer be compatible to Eagle, making Eagle->LibrePCB converters more complicated. In addition, it would still be incompatible to KiCad as well, so KiCad->LibrePCB converters still need conversion logic.
- The statement that the rotation direction currently depends on mirroring is only true when viewing at the PCB **from the top** - when viewing from the bottom to the PCB, rotation of bottom devices currently feels intuitive (a 10° rotated bottom device looks identical to a 10° rotated top device since in real life you cannot look through the PCB; you only see the devices of the board side you are looking at). When applying mirroring first, rotation of bottom devices looks inverted when looking from the bottom to the PCB.
- Unfortunately, the transformation order is a very fundamental thing of a CAD :see_no_evil: So it would be a breaking change and we need an upgrade logic for the file format to convert all coordinates to the new concept. User-created tools working directly on the *.lp files (e.g. `librepcb-parts-generator`) might need to be updated accordingly. This transition might (temporarily) lead to some pain, new bugs and "legacy code" for the upgrade logic.

My suggestion how to proceed with this: Implement the new behavior (maybe quick&dirty first) to see if it really simplifies things as expected. If it looks good, let's apply this change in LibrePCB 0.2.0. Better do it now than some day later...

Any comments welcome! :slightly_smiling_face:

dbrgn, 2022-03-04:
If it makes things easier in the long run, I'd definitely do a breaking change now. The longer we'd wait, the more painful it gets.

Regarding intuition: I find it unintuitive that rotation is CCW and not CW 🙈 What's the reason for this?

ubruhin, 2022-03-04:
> Regarding intuition: I find it unintuitive that rotation is CCW and not CW see_no_evil What's the reason for this?

CCW is basically the de-facto standard in the CAD world and mathematics world :wink: See for example the [unit circle](https://en.wikipedia.org/wiki/Unit_circle), [polar coordinate system](https://en.wikipedia.org/wiki/Polar_coordinate_system) or [cartesian coordinate system](https://en.wikipedia.org/wiki/Cartesian_coordinate_system#Rotation), all specify CCW as rotation direction. Even the Wikipedia article about [rotation](https://en.wikipedia.org/wiki/Rotation_(mathematics)) states:

> a clockwise rotation is a negative magnitude so a counterclockwise turn has a positive magnitude.

ubruhin, 2022-04-15:
TODO: Also check how auto-rotation of texts should work. Currently texts on the bottom side are auto-rotated consistently to top-side texts in terms of appearance on screen, but not in terms of when looking on the real PCB (i.e. looking from the bottom side). Maybe this behavior should also be adjusted when changing the mirror/rotate behavior.

i9e1, 2022-04-19:
regarding the KiCad mirroring:
I assume they had symbols in mind in the first place. Consider OpAmps, some draw the inv above the non-inv input and the other way around. But usually you wont draw a signal flow from right to left. Or at least feels unintuitive for me. Never mind.

But what about setting the mirror axis as user config? This should be just UI, depending on the flag add a rotation or not.

PR #1152 (2023-04-17): Change transformation order of mirror and rotate
Changing the transformation order from *rotate -> mirror -> translate* to *mirror -> rotate -> translate*. See #956 for details. I think this way the transformations are indeed more intuitive and at least slightly simpler to implement.

The `rotation` parameter of affected objects are automatically converted to the new transformation order during the file format upgrade to keep the positioning of all board/schematic items.

Closes #956