LibrePCB/LibrePCB #972: M1 Mac Apple Silicon build.

halfshavedyaks, 2022-04-23:
Is there a apple silicon build of LibrePCB?

Are there plans for one?

It's working on my M1 mac but there are some small glitchy behaviour and delays, not sure if they are related to it being translated or not.

If no one else more experineced is able to build for Apople Silicon, I'd be willing to have a go, but I would probably need help and advice.

ubruhin, 2022-04-23:
There are currently no M1 builds because we don't have access to M1 hardware and so far Azure Pipelines (where we build the official binaries) doesn't provide a build agent for M1...

halfshavedyaks, 2022-04-23:
Ok, presumably Azure will have to add that at some point since all new macs will need it.

In the meantime, If it is of interest I'm willing to make an unofficial binary on my M1 mac and make it available for download - but someone would need to talk me through it. I have some experience with writing and compiling software, but not a lot, and on mac I've mostly used Xcode. 

I think good mac support is pretty important as there is very little usable software for makers of any sort on mac.

ubruhin, 2022-04-23:
Yeah I hope Azure will add it, but I suspect this won't happen in near future since this is probably not just a simple software update.

If you want to give it a try building LibrePCB on a M1, I'd recommend to check out these resources:
- Our build instructions: https://developers.librepcb.org/d5/d96/doc_building.html
- CI build config for MacOS, containing all the steps currently done to build our official MacOS binaries: https://github.com/LibrePCB/LibrePCB/blob/master/ci/azure-jobs-macos.yml

> I think good mac support is pretty important as there is very little usable software for makers of any sort on mac.

Absolutely, it is one of the major goals of LibrePCB to provide decent PCB design software for *everyone*. Unfortunately both Apple and Microsoft make it very hard to build & test software on their OS's using CI (or even just locally) without paying lots of money for hardware, software licenses or signing keys :sob:

halfshavedyaks, 2022-04-24:
I had a go in both cmake and with qtcreator I got the same error in both.

```
CMake Error at CMakeLists.txt:100 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" (requested
  version 5.5.0) with any of the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.
```

I'm sure this is simple to fix, but it is a little over my head right now... I'll come back to it. any pointers welcome.

dbrgn, 2022-04-24:
It seems you are missing Qt 5 development headers. Maybe https://doc.qt.io/qt-5/macos.html helps, or maybe it can be installed through Homebrew somehow.

halfshavedyaks, 2022-04-24:
The example project "coffee" in Qt creator runs - so I think that everything Qt needs must be there, the problem is connecting the librePCB project to Qt somehow.

There's a lot of terminology I have only the vaguest idea about.

eg how do I "Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR" to a directory containing one of the above files."

Sorry for the noob questions, but I just don't have enough contextual knowledge to understand the instructions well enough.

dbrgn, 2022-04-24:
I think you can pass in `-DCMAKE_PREFIX_PATH=path1;path2;path3` when invoking cmake. This should generate an updated build config.

halfshavedyaks, 2022-04-24:
right! I searched for the file and found it buried many layers deep in the Qt folder - I then pasted that path into the QT Creator field that seemed right for it.

This got me a lot further, and  a lot of stuff compiled.

however it stopped with this message and a lot more of the same after it. 

so, what does "Undefined symbols for architecture arm64" actually mean? is it a configuration problem, or something missing from librePCB? I assume the source code is architcture independent so it must be some sort of configuration issue...

```
liblibrepcb_common.a(exceptions.cpp.o):-1: error: Undefined symbols for architecture arm64:
  "qt_assert_x(char const*, char const*, char const*, int)", referenced from:
      QMutexLocker::QMutexLocker(QBasicMutex*) in liblibrepcb_common.a(debug.cpp.o)
  "qFlagLocation(char const*)", referenced from:
      MainWindow::MainWindow(QWidget*) in mainwindow.cpp.o
      Ui_MainWindow::setupUi(QMainWindow*) in mainwindow.cpp.o
  "qInstallMessageHandler(void (*)(QtMsgType, QMessageLogContext const&, QString const&))", referenced from:
      librepcb::Debug::Debug() in liblibrepcb_common.a(debug.cpp.o)
  "qrand()", referenced from:
      librepcb::FilePath::getRandomTempPath() in liblibrepcb_common.a(filepath.cpp.o)
  "qt_assert(char const*, char const*, int)", referenced from:
      QString::QString(QString const&) in mainwindow.cpp.o
      QString::operator[](int) const in liblibrepcb_common.a(uuid.cpp.o)
      QArrayData::data() in liblibrepcb_common.a(uuid.cpp.o)
      QString::QString(QString const&) in liblibrepcb_common.a(uuid.cpp.o)
      QString::QString(QString const&) in liblibrepcb_common.a(exceptions.cpp.o)
      QArrayData::data() in liblibrepcb_common.a(exceptions.cpp.o)
      librepcb::Debug::messageHandler(QtMsgType, QMessageLogContext const&, QString const&) in liblibrepcb_common.a(debug.cpp.o)
```

halfshavedyaks, 2022-05-06:
I decided my cmake and Qt5 instalations were not right so I remove dthem and started over installing from brew as per the instructions in the readme.

the instruction `brew unlink qt && brew link --force qt5` returned the error `Error: No such keg: /opt/homebrew/Cellar/qt`

Nevertheless, after following various setup suggestions in the Qt installer libre PCB compiled natively and it runs well, and faster than the intel version.

However the built .app package file is not a self contained mac app, it only runs if run from the location (build directory) where it was compiled. even moving the entire librepcb folder breaks it. So I guess something is configured with relative paths. 

Also the .app package is a suspiciously small 9.7MB - seems like maybe it is relying on some things that are maybe in the librepcb folder but not incorporated in the mac app package.

Does anyone know how to make it compile everything it needs into the app so it can be copied to any location?

halfshavedyaks, 2022-05-06:
BTW I compiled the release version - not the github master

ubruhin, 2022-05-06:
Great! :smiley: 

For our official binaries, we use [macdeployqt](https://doc.qt.io/qt-5/macos-deployment.html#macdeploy) to make it self-contained.

Generally you need to compile LibrePCB [as done on our CI](https://github.com/LibrePCB/LibrePCB/blob/master/ci/build_librepcb.sh):

    mkdir -p build && cd build
    cmake .. -DCMAKE_INSTALL_PREFIX=$(pwd)/install/opt
    make -j8
    make install

And afterwards call [`./ci/build_mac_bundle.sh`](https://github.com/LibrePCB/LibrePCB/blob/master/ci/build_mac_bundle.sh) to build the self-contained *.dmg bundles. Note that this file is not part of our source releases, but you can just execute the commands manually in your terminal instead of executing the whole script:

```
mv "./build/install/opt/bin/librepcb.app" "./build/install/opt/librepcb.app"
cp -r "./build/install/opt/share" "./build/install/opt/librepcb.app/Contents/"
mv "./build/install/opt/bin/librepcb-cli.app" "./build/install/opt/librepcb-cli.app"
cp -r "./build/install/opt/share" "./build/install/opt/librepcb-cli.app/Contents/"
rm -r "./build/install/opt/bin" "./build/install/opt/share"
macdeployqt "./build/install/opt/librepcb.app" -dmg
mv ./build/install/opt/librepcb.dmg ./librepcb.dmg
macdeployqt "./build/install/opt/librepcb-cli.app" -dmg
mv ./build/install/opt/librepcb-cli.dmg ./librepcb-cli.dmg
```

halfshavedyaks, 2022-05-06:
great thanks!

This builds an app and a dmg with an app in it that are a plausible size, however when trying to run it I get an librePCB quit unexpectedly message.

the report contains this:

`Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x00000001051b8000
Exception Codes:       0x0000000000000032, 0x00000001051b8000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace CODESIGNING, Code 2 `

so a codesigning issue?

I did some googling but the fixes mentioned are over my head.

halfshavedyaks, 2022-05-06:
After running cmake .. -DCMAKE_INSTALL_PREFIX=$(pwd)/install/opt

I got this warning:

```
-- Could NOT find Qt5Zlib (missing: Qt5Zlib_DIR)
CMake Warning at /opt/homebrew/opt/qt@5/lib/cmake/Qt5/Qt5Config.cmake:39 (message):
  Failed to find Qt5 component "Zlib" config file at
  "/opt/homebrew/opt/qtmodulemodule@Config.cmake"
Call Stack (most recent call first):
  libs/quazip/CMakeLists.txt:67 (find_package)
```
does this matter?

halfshavedyaks, 2022-05-06:
i don't know what changed but i stopped even getting the  error, now the app just fails to start with no messages.
i tried starting over and rebuilding everything , same result, no error but won't start.

however it runs fine from /librepcb-0.1.6/build/apps/librepcb/librepcb.app

ubruhin, 2022-05-25:
Hm okay then I don't have any idea what could be the problem... The only thing I can say is that the CMake warning about Qt5 is not relevant, I get this warning as well :wink:

ubruhin, 2022-10-02:
For the record: I saw that macOS generally supports multi-arch binaries and it [looks quite easy](https://www.qt.io/blog/qt-on-apple-silicon) to build them on a x86_64 macOS (i.e. the one we have on CI). I gave it a try, but unfortunately [Homebrew does not allow to install multi-arch Qt libraries](https://github.com/Homebrew/discussions/discussions/1609) so it's not that easy to get it running on our CI. I don't know if there's another way to install Qt with multi-arch support...

If multi-arch Qt libraries are installed, building multi-arch LibrePCB binaries should be as easy as passing `-DCMAKE_OSX_ARCHITECTURES="x86_64;arm64"` to CMake.

ubruhin, 2023-08-08:
Removed this from the 1.0.0 milestone because I see no way to provide M1 builds at the moment. From my point of view, there are only two variants to fix this issue:

- Hope and wait for Azure Pipelines to add free M1 runners (this could take years).
- Someone sponsoring an M1 device so I can build LibrePCB releases manually on that device.

I think building the binaries by someone else is not really an option for official release as there's no way to proof the integrity of the binaries :-/

dbrgn, 2023-08-24:
@ubruhin on M1, in the long run, you won't have a good UX without signing and notarizing the binary. This was already hard on Intel Macs, but on Apple Silicon they seem to have increased the requirements.

ubruhin, 2023-08-24:
> @ubruhin on M1, in the long run, you won't have a good UX without signing and notarizing the binary. This was already hard on Intel Macs, but on Apple Silicon they seem to have increased the requirements.

Yeah I noticed that, it took me quite some time to get the binaries running (see #1209). But I have no idea how we should proceed to fix the horrible user experience... I don't like to pay Apple money to get an open-source application published. Also I don't like to spend many many more hours on this topic, everything related to macOS feels so cumbersome to me as a non-mac user. And building the binaries manually every time (due to the absence of an arm64 CI runner) is cumbersome as well.

halfshavedyaks, 2023-08-29:
That's great! thanks! I have downloaded and it runs, I can see my files.

getting it running isn't so hard for someone used to running unsigned software on a mac. We get used to defeating apple's silly ideas.

I use a mac for some things and linux for others. Generally I find I have to spend significant time getting linux to work as intended  - on a mac on the other hand I have to spend a lesser, but still significant amount of time stopping it from working as apple intended!

Anyway it is great to have LibrePCB working natively on both!

M1 mac minis are getting old now and can be found more cheaply - for the future I wonder if we could crowdfund getting you one to build on?

bcr, 2023-10-01:
>     * Hope and wait for Azure Pipelines to add free M1 runners (this could take years).

Can you talk more about this? I see arm64 listed on [Self-hosted macOS agents](https://learn.microsoft.com/en-us/azure/devops/pipelines/agents/osx-agent) -- is they key word here "free?" This may be something we can escalate with Microsoft if that's the case.

This may get to be an existential concern for the Mac builds (like you may have to drop Mac as a supported platform.) When we did the original PowerPC -> Intel transition, Apple only provided Rosetta for five years. For this Intel -> Apple Silicon transition, Rosetta 2 came out in 2020 and this time is likely going to run out in 2025.

I don't know if you have telemetry about what platforms people are running LibrePCB on. You may be able to collect this data anonymously through User-Agent data when downloading parts libraries, and see if there's enough Mac users to warrant the effort of supporting the platform.

ubruhin, 2023-10-01:
> Can you talk more about this? I see arm64 listed on [Self-hosted macOS agents](https://learn.microsoft.com/en-us/azure/devops/pipelines/agents/osx-agent)

The problem is the term *self-hosted* - it means we would need to setup and run our own CI runner on our own (or rented) Apple Silicon hardware. So this does not help at all. What we need is a *hosted* runner.

Btw, in the mean time I got a fair offer for a M1 Mac Mini which I have now bought to create the M1 build for LibrePCB 1.0 locally (any donations from Apple Silicon users appreciated! :slightly_smiling_face: I bought it specifically for you).

It's not an ideal solution since it involves additional manual release effort and makes releasing only possible with access to this device (no releases from remote work :grin:) but at least I have now a way to build for Apple Silicon.

Since the original request of this issue (providing an M1 build) is now solved, I'll close it. But feel free to continue the discussion here. And of course the goal is still to get an arm64 CI runner.

closed 2023-10-01 by ubruhin

nimish, 2023-12-26:
macOS runners on Apple Silicon are now available: https://github.com/actions/runner-images/issues/8439

radeva, 2024-01-18:
> The problem is the term self-hosted - it means we would need to setup and run our own CI runner on our own (or rented) Apple Silicon hardware.

@ubruhin You can try [FlyCI](https://www.flyci.net/)'s M1 and M2 runners. They are on average 2x faster and 2x cheaper than GitHub's **AND** we have a free tier for OSS projects (see below).

# Install Instructions

1. Install [FlyCI app](https://github.com/apps/flyci-prod) and 
2. Easily replace one one and start using FlyCI runners:

```diff
jobs:
 ci:
-    runs-on: macos-latest
+    runs-on: flyci-macos-large-latest-m1
   steps:
   - name: 👀 Checkout repo
     uses: actions/checkout@v4
```
# 500 mins/month Free for Public Repos

Since your repo is public, FlyCI offers 500 mins/month of free M1 runner usage with the `flyci-macos-large-latest-m1` runner for public projects. 

Don't hesitate to contact us in case the free tier doesn't suit your needs or you experience any issues with the runners. Our team is here to support you!

Best Regards,
Veselina Radeva
Product Manager at [FlyCI](https://flyci.net/)