LibrePCB/LibrePCB #974: Board editor: Top and bottom planes can't be fully filled. In most cases bottom plane can't be filled at all for some reasons.

azhizhinov, 2022-05-01:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.6
Git Revision: c4f316428
Qt Version: 5.15.3 (built against 5.15.3)
Operating System: Gentoo/Linux
...
```

##### SUMMARY

Top and bottom planes can't be fully filled. In most cases bottom plane can't be filled at all for some reasons.

##### STEPS TO REPRODUCE

Project .lppz has been attached.

##### EXPECTED RESULTS

Planes are filled.

##### ACTUAL RESULTS

Top plane has been filled partially. Bottom plane not filled at all.

##### ADDITIONAL INFORMATION

[image]
[image]
[For_Science_Keyboard.lppz.zip](https://github.com/LibrePCB/LibrePCB/files/8598504/For_Science_Keyboard.lppz.zip)

ubruhin, 2022-05-25:
Thanks for the detailed bug report. However, I cannot reproduce it :see_no_evil: 

Do you know whether your LibrePCB binary was built with the bundled polyclipping library or with polyclipping from the package manager (using `UNBUNDLE_POLYCLIPPING`)? If unbundled, what version was used?

Alternatively you could try to reproduce it with our official AppImage.

ktrace, 2022-05-27:
-DUNBUNDLE_QUAZIP=ON
-DUNBUNDLE_MUPARSER=ON
Bundled polyclipping. I use same ebuild but can't reproduce too. Need more info.

azhizhinov, 2022-06-12:
Hello.
Sorry for delay.
I'm not experienced enough to debug this issue, sorry.
So I have 2 laptops with the same hardware and this issue is reproducible always.
Both releases with the same issue.
Can this issue be hardware dependent somehow?
LibrePCB is built from sources.
https://mirror.yandex.ru/gentoo-distfiles/distfiles/librepcb-0.1.5-source.zip
https://mirror.yandex.ru/gentoo-distfiles/distfiles/librepcb-0.1.6-source.zip
Resulting packages are.
[librepcb-0.1.5.tbz2.zip](https://github.com/LibrePCB/LibrePCB/files/8885864/librepcb-0.1.5.tbz2.zip)
[librepcb-0.1.6.tbz2.zip](https://github.com/LibrePCB/LibrePCB/files/8885873/librepcb-0.1.6.tbz2.zip)
[image]
[image]
[image]

azhizhinov, 2022-06-14:
Hello.
Big thanks for Viktor K (ktrace).
The reason was found.
Both releases were built with these flags:
```
Compiler flags:
C -march=haswell -O2 -pipe
C++ -march=haswell -O2 -pipe
```
Today, as Victor K advised, I've re-compiled 0.1.6 release with these flags:
```
C -O2 -march=x86-64 -pipe
C++ -O2 -march=x86-64 -pipe
```
And now plates are filled as expected. Yay.
So, your code is very fragile.
[image]
[image]
[image]

ubruhin, 2022-07-31:
Awesome, thanks a lot for providing these details! With `-march=haswell` I was able to reproduce the bug and after quite some debugging I saw that different floating point optimizations lead to a slightly negative number instead of zero, which was passed to `qSqrt()` and then messed up the plane calculations :sob:

Fixed it in #991 :rocket:

azhizhinov, 2022-07-31:
Hello.That's awesome, thanks a lot.

PR #991 (2022-07-31): Fix potential numeric issues with "-march=haswell"
Due to potential numeric inaccuracy which was observed with "-march=haswell" (see #974), the argument passed to `qSqrt()` could be a tiny negative number instead of zero, which leads to NaN return value and thus various problems, for example broken plane fragment calculations. Probably it is caused by floating-point optimizations. Fixed it by clipping negative numbers to zero.

Fixes #974
closed 2022-07-31 by ubruhin