LibrePCB/LibrePCB #977: OSX Dialog boxes don't match manual

Ben01880, 2022-05-02:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version:
Git Revision:
Qt Version:
Operating System:
...
```

##### SUMMARY

##### STEPS TO REPRODUCE

##### EXPECTED RESULTS

##### ACTUAL RESULTS

##### ADDITIONAL INFORMATION

dbrgn, 2022-05-03:
@Ben01880 we need a bit more detail than that. Which dialog boxes are you talking about?

Ben01880, 2022-05-03:
I download and installed LibrePCB
OSX Monterey and Ubuntu Budgie
Downloaded the Manual.  ( Does it have a Version Number ? )
The manual appears to be showing OSX style window/dialogs.
Both versions don’t seem to be visually equal.
Is it me or are there  Back, Finish, Cancel buttons missing.

ubruhin, 2023-02-21:
Most (or even all) screenshots were updated to the latest LibrePCB version in https://github.com/LibrePCB/librepcb-doc/pull/66, so they should not be outdated anymore.

However, the screenshots are made on Ubuntu. Since dialog buttons depend on the operating system, they might be different on macOS. This is not a bug (it makes no sense to write a separate manual for each operating system).

So let's close this issue.

closed 2023-02-21 by ubruhin